Guilouz/Creality-Helper-Script-Wiki #116: [BUG] <title> start input shaper if i puase the print

antisocial77, 2023-11-18:
### What happened?

i just rooted the k1 and realize that when i pause the print, input shaping starts.

### What did you expect to happen instead?

i just want to pause without input shaping :)

### How to reproduce this issue?

when i pause the print, input shaping starts.

Guilouz, 2023-11-18:
Input shaping is the resonnance test, it is never launched during print or during a pause.

Make sur you have correct configuration files.

antisocial77, 2023-11-18:
yes i know (even im kinda noob) but it happened 4 times. here is my printer.cfg
[config-20231118-184534.zip](https://github.com/Guilouz/Creality-K1-and-K1-Max/files/13400751/config-20231118-184534.zip)

Guilouz, 2023-11-18:
That's macros who pilot printer, tou just send printer.cfg.

closed 2023-11-27 by Guilouz