Guilouz/Creality-Helper-Script-Wiki #120: octoeverywhere wont uninstall[BUG] <title>

mindkill666, 2023-11-19:
### What happened?

I wanted to uninstall this and it is still there.
I installed newest script version and uninstalled it, after checking i could still uninstall it again and again.
Is there a way i can manually do this?

### What did you expect to happen instead?

That it would be gone.

### How to reproduce this issue?

don't know

Guilouz, 2023-11-19:
You can just remove octoeverywhere folder in /usr/share folder.

@QuinnDamerell must send email to user to this since recent changes in install folder.

QuinnDamerell, 2023-11-19:
Yes! I was going to send the emails Monday. I can also update the removal system in the uninstall script to special case that old folder and delete it.

mindkill666, 2023-11-19:
Problem is that the files are in the data folder, not in share.

Guilouz, 2023-11-19:
What do you want to do excatly?

mindkill666, 2023-11-19:
uninstall it, don't have use for it, but after uninstalling it still says it's installed. i tried to delete the folder there but then it gives an error in the update menu, so i probably have to delelte some other files.

Guilouz, 2023-11-19:
Maybe an issue in uninstall script of octoeverywhere. @QuinnDamerell can you check that please.

Guilouz, 2023-11-19:
@QuinnDamerell in your uninstall.sh script that return to: sh ./install.sh -uninstall

But I didn't see any function named uninstall in your install.sh.

QuinnDamerell, 2023-11-19:
Hey! I'm working on it now, I should be done in just a few minutes. The -uninstall flag is handled by the PY installer component

QuinnDamerell, 2023-11-19:
Actually I just finished, I'm doing some testing fully uninstalling and re-installed moonraker/fluidd/octoeverywhere and it's looking good so far, but I will report back when I'm fully done.

QuinnDamerell, 2023-11-19:
Yup, it's all looking great! The uninstall will remove all of the files, remove the moonraker config, stop the services, and cleanup any installs in /usr/share.

Let me know if you see any more issues!

closed 2023-11-20 by Guilouz

Guilouz, 2023-11-21:
@QuinnDamerell Just installed octoeverywhere, it's look like octoeverywhere is always installed in /usr/share/ and octoeverywhere-env in /usr/data/

so it's still not correct, octoeverywhere must be installed in /usr/data/