Guilouz/Creality-Helper-Script-Wiki #124: [BUG] Add support for re-installing features after firmware update

evgarthub, 2023-11-20:
### What happened?

After firmware update features are gone due to missing parts but installer script tells you that feature is installed. 

### What did you expect to happen instead?

It would be nice to install missing dependencies for such cases instead of going through and uninstalling-installing features one by one.

### How to reproduce this issue?

1. Install KAMP/Moonraker/Mainsail etc
2. Update printer firmware
3. Re-configure printer.cfg and macros
4. Features are not working or showing errors due to missing dependencies
5. When running installer script again install option warns you that feature is already installed.

### Additional information:

Nice to have as a script option or just add step by step guide for future firmware updates.

Guilouz, 2023-11-20:
After every update you must restore default settings to avoid issue by following this : https://github.com/Guilouz/Creality-K1-and-K1-Max/wiki/Reset-Factory-Settings

So all features must be reinstalled.

closed 2023-11-20 by Guilouz

evgarthub, 2023-11-21:
Great! Thank you for pointing out!