Guilouz/Creality-Helper-Script-Wiki #128: [BUG] Difference in Documentation and Values for Improve Hotend Cooling...

DanBegallieHW, 2023-11-21:
### What happened?

In the Improve hotend cooling at the end of printing, the text says:

### Improve hotend cooling at the end of printing
By default at the end of a print, the hotend is cooled by its fan which runs at 100%.

By making this change, the hotend fan speed will be reduced to 80% and the side fan will also be enabled at 40%. This allows faster and quieter cooling.

But the macro code sets both fan values to 210 which is about 82% for both fans

```
[gcode_macro WAIT_TEMP_START]
gcode:
  UPDATE_DELAYED_GCODE ID=wait_temp DURATION=1
  SET_PIN PIN=fan0 VALUE=210
  SET_PIN PIN=fan2 VALUE=210
```

### What did you expect to happen instead?

Have the fans set to 80% and 40% respectively or change the documentation if the values are correct.

```[gcode_macro WAIT_TEMP_START]
gcode:
  UPDATE_DELAYED_GCODE ID=wait_temp DURATION=1
  SET_PIN PIN=fan0 VALUE=204
  SET_PIN PIN=fan2 VALUE=102
```

### How to reproduce this issue?

Watch at end of print for fan values.

Guilouz, 2023-11-21:
Nope, fan0 and fan2 have not the same settings, so with value 210 fan0 work at 60% and fan2 at 40%

closed 2023-11-21 by Guilouz

DanBegallieHW, 2023-11-22:
Thank you for the explanation.  Definitely not aware.