Guilouz/Creality-Helper-Script-Wiki #131: Fluidd fails to install

jacquesglerner, 2023-11-22:
### What happened?

I started Helper and installed Moonraker and Nginx, ok #`1. Then tried to install Fluidd, it goes normal to the install routine. I tried Fluidd option #2 and y (yes)

### What did you expect to happen instead?

Fluidd to be installed. I get 1 or 2 seconds and return to the Main Install page. WHen I try to see what is installed I get that Fluidd was not installed.

### How to reproduce this issue?

Just redid this routine several times, including a reboot at no avail.

### Additional information:

FW is 1.3.2.20 It was working on 1.3.2.1 Printer is K1.

jacquesglerner, 2023-11-22:
Just for your information http:<myIP>:7125 is OK, no remarks!

jacquesglerner, 2023-11-22:
[image]

joschiv1977, 2023-11-22:
If you have already installed Moonraker, Fluidd or Mainsail provided by Creality, please uninstall them first using [Remove Menu].

jacquesglerner, 2023-11-22:
Nope! Always Fluidd from this Github. Just plain Guilouz. By the way, my only update from 1.3.2.1 to 1.3.2.20 was only made via air because its public (for all).

Guilouz, 2023-11-22:
did you do a factory reset after installing the new firmware?

jacquesglerner, 2023-11-22:
> did you do a factory reset after installing the new firmware?

No> But did afterwords before installing Moonraker  (and Fluidd). Result is That It's ok for Moonraker and no good for Fluidd.

brunobarretosa, 2023-11-23:
Same here :(
Did factory reset already using the 1.3.2.2 and can't install Fluidd nor Mailsail.

error:
"wget: TLS error from peer (alert code 80): 80
wget: error getting response: Connection reset by peer
 Download failed. Exit code: 1"

brunobarretosa, 2023-11-23:
Same here. Just saw there's no Fluidd / Mainsail folder into the Scripts/Files folders. Is the installer script trying to find them into this repository?

Guilouz, 2023-11-23:
@brunobarretosa Read the wiki for this error…

brunobarretosa, 2023-11-23:
> @brunobarretosa Read the wiki for this error…

Couldn't find. Can you please tell me where?

jacquesglerner, 2023-11-23:
> > did you do a factory reset after installing the new firmware?
> 
> No. But I did it afterward before installing Moonraker (and Fluidd). The result is that It's okay for Moonraker and no good for Fluidd.

Just adding:
Did a full factory reset, shut down the printer, and restarted from zero. 
Checked the FW # and so on...
I started pinging the K1's IP and tried to connect to the MB.
Now I have no access to SSH.

The full transcript is:

C:\Users\jgler>ssh -p 22 root@192.168.x.y
The authenticity of host '192.168.x.y (192.168.x.y)' can't be established.
ECDSA key fingerprint is SHA256:xkLEACT4vBvkhaZwVM4bdLzzt1Clu1tIsMS3dk6/YoY.
This key is not known by any other names
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added '192.168.x.y' (ECDSA) to the list of known hosts.

C:\Users\jgler>ssh -p 22 root@192.x.y
ssh: connect to host 192.168.x.y  port 22: Connection timed out

I have access to the normal page at http://192.168.x.y but no access to SSH, PuTTy, OpenSSH, etc.
That's where I am.

Guilouz, 2023-11-23:
> > @brunobarretosa Read the wiki for this error…
> 
> Thank you man! Sorry, I didn't see it before.

https://github.com/Guilouz/Creality-K1-and-K1-Max/wiki/Installation-Helper-Script

Guilouz, 2023-11-23:
> > > did you do a factory reset after installing the new firmware?
> > 
> > 
> > No. But I did it afterward before installing Moonraker (and Fluidd). The result is that It's okay for Moonraker and no good for Fluidd.
> 
> Just adding: Did a full factory reset, shut down the printer, and restarted from zero. Checked the FW # and so on... I started pinging the K1's IP and tried to connect to the MB. Now I have no access to SSH.
> 
> The full transcript is:
> 
> C:\Users\jgler>ssh -p 22 [root@192.168.x.y](mailto:root@192.168.x.y) The authenticity of host '192.168.x.y (192.168.x.y)' can't be established. ECDSA key fingerprint is SHA256:xkLEACT4vBvkhaZwVM4bdLzzt1Clu1tIsMS3dk6/YoY. This key is not known by any other names Are you sure you want to continue connecting (yes/no/[fingerprint])? yes Warning: Permanently added '192.168.x.y' (ECDSA) to the list of known hosts.
> 
> C:\Users\jgler>ssh -p 22 [root@192.x.y](mailto:root@192.x.y) ssh: connect to host 192.168.x.y port 22: Connection timed out
> 
> I have access to the normal page at http://192.168.x.y but no access to SSH, PuTTy, OpenSSH, etc. That's where I am.

It's an issue with your computer with SSH key, not related to the script.

closed 2023-11-23 by Guilouz