Guilouz/Creality-Helper-Script-Wiki #132: [BUG] Underscore before PURGE_LINE and SMART_PARK

Remleyfpv, 2023-11-23:
### What happened?

smart park and underscore didnt worki after following KAMP guide

### What did you expect to happen instead?

to smart park and purge filament before printing

### How to reproduce this issue?

Just follow the guide

### Additional information:

The FIX is to remove the under score _LINE_PURGE -> LINE_PURGE and _SMART_PARK -> SMART_PARK so please fix this in the guide for future users

Guilouz, 2023-11-23:
Underscrore is correct, you must use latest KAMP version for that. 

Underscrore in Klipper macro used to hide macro from list.

closed 2023-11-23 by Guilouz

DrFailbucket, 2023-11-27:
After Updating the software line_purge and smart_park didn't worked for me, after removing the underlines and NOT reinstall KAMP it worked just fine again