Guilouz/Creality-Helper-Script-Wiki #136: "screws_tilt_adjust" not implemented?

DrFailbucket, 2023-11-24:
### What happened?

I was trying to add the "screws_tilt_adjust" to the printer.cfg, after Rebooting it reports with this error: `{"code": "key341", "msg": "Section 'screws_tilt_adjust' is not a valid config section\n\nOnce the underlying issue is corrected, use the \"RESTART\"\ncommand to reload the config and restart the host software.\nPrinter is halted\n", "values": ["screws_tilt_adjust"]}`

### What did you expect to happen instead?

That *screws_tilt_adjust* would work as known on my other Printers

### How to reproduce this issue?

Adding `screws_tilt_adjust` to the printer.cfg and hit *Save&Restart*

### Additional information:

Comment from Official Klipper Discord: *Doesnt matter if it's rooted or not it's still creality's version of klipper, and if you get that error message it means they have removed the module that provided that particular section*

Guilouz, 2023-11-24:
Not related to the wiki. Can’t help you.

closed 2023-11-24 by Guilouz

StofflHD, 2023-11-27:
With this you can add the screw_tilt_adjust functionality. SSH into your printer and run these commands:

cd /usr/share/klipper/klippy/extras

wget -O bed_screws.py "https://raw.githubusercontent.com/CrealityOfficial/K1_Series_Klipper/main/klippy/extras/bed_screws.py"

wget -O screws_tilt_adjust.py "https://raw.githubusercontent.com/CrealityOfficial/K1_Series_Klipper/main/klippy/extras/screws_tilt_adjust.py"

DrFailbucket, 2023-11-27:
Thanks StoffelHD, i'll try it and update when i got some results 👍

DrFailbucket, 2023-11-28:
> With this you can add the screw_tilt_adjust functionality. SSH into your printer and run these commands:
>
> cd /usr/share/klipper/klippy/extras
>
> wget -O bed_screws.py [...]

THANK YOU!

Worked like a charm!