Guilouz/Creality-Helper-Script-Wiki #139: [BUG] Wiki script for heat chamber using M191 never ends

mulder999, 2023-11-25:
### What happened?

Example list 2 sensors:
      TEMPERATURE_WAIT SENSOR="~~temperature_fan~~ chamber_fan" MINIMUM={s-1} MAXIMUM={s+1}

### What did you expect to happen instead?

Correct line should read:
      TEMPERATURE_WAIT SENSOR="chamber_fan" MINIMUM={s-1} MAXIMUM={s+1}

### How to reproduce this issue?

Assuming actual chamber temperature is 25°:
- Issue `M191 S26` in gcode console
- Wait until temperature reach 26
Heating continues forever

### Additional information:

Unsure if and how I can submit a pull request for github wiki.

Guilouz, 2023-11-26:
This is not correct.

'chamber_fan' is unknown sensor so this not working :

TEMPERATURE_WAIT SENSOR="chamber_fan" MINIMUM={s-1} MAXIMUM={s+1}

M191 Macro use this:

[temperature_fan chamber_fan]
pin: PC0
cycle_time: 0.0100
hardware_pwm: false
max_power: 1
shutdown_speed: 0
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC5
min_temp: 0
max_temp: 70
control: watermark
max_delta: 2
target_temp: 35.0
max_speed: 1.0
min_speed: 0.0

mulder999, 2023-11-26:
Thanks a lot for answer. I implemented exactly the wiki content including this definition. Heating of the bed never stops. Unsure what the actual error is but something is wrong on the end condition.

Guilouz, 2023-11-26:
The M191 command is intended to be launched before a print, in particular a gcode action of the slicer. Here the goal is to raise the room to a target temperature by activating the bed at 100C before starting printing which has a temperature setpoint for the bed so the heating does not need to be changed since modified by the gcode before printing.

Guilouz, 2023-11-27:
I try but i don’t understand what is the interest of the last command M140 S{s}.

When chamber temperature is reached print start and bed temp is given by slicer.

mulder999, 2023-11-27:
I fully understand your assumption, it may be valid under some circumstances, the code is just more robust with this addition (slicer configuration issues if START_PRINT arises before M191) and also opens up for theses alternative scenarios:
- Preheating the chamber in advance by the user
- Drying filament in the printer

Nevertheless the major issue with the code was that it was setting a MAXIMUM value in TEMPERATURE_WAIT causing infinite heating and wait when the temperature was already above the max at the time the M191 command was issued.

closed 2023-11-27 by Guilouz

Dashrendar2507, 2024-01-06:
> Well I believe it should not overshoot the value by blindly assuming a proper gcode will immediately follow.
> 
> Below is an improved version:
> 
> ```
> [gcode_macro M191]
> description: Wait Chamber Temperature To Heat Up
> gcode:
>   {% set s = params.S|float %}
>   {% set chamber_temp = printer["temperature_sensor chamber_temp"].temperature|float %}
>   {% if s > 0 %}
>     M141 S{s}
>   {% endif %}
>   {% if s > chamber_temp and s <= 90 %}
>     M140 S100
>     { action_respond_info("Waiting for the bed to heat up the chamber...") }
>     TEMPERATURE_WAIT SENSOR="temperature_fan chamber_fan" MINIMUM={s-1}
>     { action_respond_info("Chamber target temperature reached: %s°C" % (s)) }
>     M140 S{s}
>   {% endif %}
> ```
> 
> Changes are:
> 
> * Setup a bed value for the long run if bed temperature is not changed after a M191 command
> * Do not heat bed and wait if the chamber temperature is already reached as the condition would likely never occurs
> * Remove the maximum temperature limit check as it can serve only during cooling phase and might block forever (see additional improvement above)
> * Reuse macro M141 to avoid rewrite
> * Limit upper bound to 90°
> * Improve wording in respond_info

There's something I don't understand about the " if s > chamber_temp and s <= 90 %". On mine, the "Chamber target temperature reached" triggers immediately because it looks at the "temperature_fan chamber_fan", which we use to set the target chamber temp as far as I understand. Then, in the "TEMPERATURE_WAIT", shouldn't we be using "temperature_sensor chamber_temp"? My question is why not use the temp sensor instead of the fan ?