Guilouz/Creality-Helper-Script-Wiki #143: [BUG] <No Purgeline on v1.1.2-6-gd4dee4c8>

Nazgile94, 2023-11-27:
### What happened?

01:17
{"code":"key61, "msg":"Unknown command:_LINE_PURGE", "values": ["_LINE_PURGE"]}
01:16
{"code":"key61, "msg":"Unknown command:_SMART_PARK", "values": ["_SMART_PARK"]}

### What did you expect to happen instead?

purgeline printed

### How to reproduce this issue?

just updated to new version

### Additional information:

already uninstalled via helper script, updated helperscript and reinstalled kamp - got the erro 01:17
{"code":"key61, "msg":"Unknown command:_LINE_PURGE", "values": ["_LINE_PURGE"]}
01:16
{"code":"key61, "msg":"Unknown command:_SMART_PARK", "values": ["_SMART_PARK"]} 

and he starts print after doing kamp bed mesh.

Guilouz, 2023-11-27:
https://github.com/Guilouz/Creality-K1-and-K1-Max/wiki/Wiki-Changelog

closed 2023-11-27 by Guilouz

Nazgile94, 2023-11-27:
Thanks, fixed 👌🥳