Guilouz/Creality-Helper-Script-Wiki #147: Timelapse In Browser viewer Issue

hjkeller, 2023-11-27:
### What happened?

When attempting to view timelapse videos via mainsail in firefox I am met with the following error:
"No video with supported format and MIME type found."
[image]
When attempting to view via the device tab of orcaslicer I am met with a blank video window.
[image]

### What did you expect to happen instead?

The expected behavior is that the timelapse clip plays back in the webui or orca slicer device tab.

### How to reproduce this issue?

Attempt to view any video in the timelapse section.

### Additional information:
[config.zip](https://github.com/Guilouz/Creality-K1-and-K1-Max/files/13479666/config.zip)

[image]

Guilouz, 2023-11-27:
Not related to the wiki. Please post only issue related to the wiki.

closed 2023-11-27 by Guilouz

hjkeller, 2023-11-27:
> Not related to the wiki. Please post only issue related to the wiki.

Please advise the appropriate location to post. Is this something you support or should I go to moonraker/timelapse directly?

Guilouz, 2023-11-27:
Camera working on another web browser?

hjkeller, 2023-11-27:
> Camera working on another web browser?

The camera works and the timelapase videos can be downloaded and viewed in VLC. Just not in the browser/slicer timelapse pane.

Guilouz, 2023-11-27:
Working fine with Fluidd. Maybe an issue with your browser or with Mainsail.

ASCONTRACT1, 2023-12-24:
> This is what happened in Fluidd: ![image](https://private-user-images.githubusercontent.com/14224724/286037637-1f3da667-8f1d-47a3-beda-f254b4f7d45e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMzOTczMzIsIm5iZiI6MTcwMzM5NzAzMiwicGF0aCI6Ii8xNDIyNDcyNC8yODYwMzc2MzctMWYzZGE2NjctOGYxZC00N2EzLWJlZGEtZjI1NGI0ZjdkNDVlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI0VDA1NTAzMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ3YzlkNTZjMGEzODQ4MTczNzZlZmFmNTVlMWE0OGRhNGQ2ZjRjYzcyNjI5YjE3OThhZGRjOWJhZjY0YTc4YmQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.BLPbDOeesI4YelSgl3pIfbx9kPA9xOndAaODhdUrCEk) Happens on firefox, mobile firefox, and chromium based browsers in fluidd or mainsail. This has happened after factory installs also. Any other thoughts or suggestions on where I might check?

Did you ever figure this out im running into the same issue where it doesnt play in browser or orca and after download gives .mp4v codecs error. My sonic pad klipper instance doesn’t have this issue with timelapse so it would be nice to fix this.

hjkeller, 2023-12-26:
> Did you ever figure this out im running into the same issue where it doesnt play in browser or orca and after download gives .mp4v codecs error. My sonic pad klipper instance doesn’t have this issue [...]

Regrettably I have not. I even did a full factory reset and couldn't get it to work.