Guilouz/Creality-Helper-Script-Wiki #150: [BUG] Input Shaper

karrenbauer, 2023-11-29:
### What happened?

get error:
!! {"code": "", "msg":"MCU 'mcu' shutdown: Move queue overflow", "values": []} 
when run INPUT_SHAPER macro on K1 (not Max)

after Line 124 Hz
// Testing frequency 124 Hz
08:23:01 
// Klipper state: Shutdown

### What did you expect to happen instead?

run makro:

### How to reproduce this issue?

[gcode_macro INPUT_SHAPER]
description: Measure X and Y axis resonances
gcode:
  G90
  {% if printer.toolhead.homed_axes != "xyz" %}
    G28
  {% endif %}
  SHAPER_CALIBRATE
  {% set y_park = printer.toolhead.axis_maximum.y/2 %}
  {% set x_park = printer.toolhead.axis_maximum.x|float - 10.0 %}
  G1 X{x_park} Y{y_park} F20000
  M84

### Additional information:

08:22:04 
// Testing frequency 82 Hz
08:22:05 
// Testing frequency 83 Hz
08:22:06 
// Testing frequency 84 Hz
08:22:08 
// Testing frequency 85 Hz
08:22:09 
// Testing frequency 86 Hz
08:22:10 
// Testing frequency 87 Hz
08:22:10 
// Testing frequency 88 Hz
08:22:12 
// Testing frequency 89 Hz
08:22:13 
// Testing frequency 90 Hz
08:22:15 
// Testing frequency 91 Hz
08:22:15 
// Testing frequency 92 Hz
08:22:16 
// Testing frequency 93 Hz
08:22:16 
// Testing frequency 94 Hz
08:22:18 
// Testing frequency 95 Hz
08:22:19 
// Testing frequency 96 Hz
08:22:20 
// Testing frequency 97 Hz
08:22:21 
// Testing frequency 98 Hz
08:22:22 
// Testing frequency 99 Hz
08:22:23 
// Testing frequency 100 Hz
08:22:23 
// Testing frequency 101 Hz
08:22:25 
// Testing frequency 102 Hz
08:22:26 
// Testing frequency 103 Hz
08:22:27 
// Testing frequency 104 Hz
08:22:28 
// Testing frequency 105 Hz
08:22:30 
// Testing frequency 106 Hz
08:22:30 
// Testing frequency 107 Hz
08:22:42 
// Testing frequency 108 Hz
08:22:51 
// Testing frequency 109 Hz
08:22:55 
// Testing frequency 110 Hz
08:22:58 
// Testing frequency 111 Hz
08:23:00 
// Testing frequency 112 Hz
08:23:00 
// Testing frequency 113 Hz
08:23:00 
// Testing frequency 114 Hz
08:23:00 
// Testing frequency 115 Hz
08:23:00 
// Testing frequency 116 Hz
08:23:01 
// Testing frequency 117 Hz
08:23:01 
// Testing frequency 118 Hz
08:23:01 
// Testing frequency 119 Hz
08:23:01 
// Testing frequency 120 Hz
08:23:01 
// Testing frequency 121 Hz
08:23:01 
// Testing frequency 122 Hz
08:23:01 
// Testing frequency 123 Hz
08:23:01 
// Testing frequency 124 Hz
08:23:01 
// Klipper state: Shutdown
08:23:01 
// Testing frequency 125 Hz
08:23:01 
// Testing frequency 126 Hz
08:23:01 
// Testing frequency 127 Hz
08:23:01 
// Testing frequency 128 Hz
08:23:01 
// Testing frequency 129 Hz
08:23:02 
// Testing frequency 130 Hz
08:23:02 
// Testing frequency 131 Hz
08:23:03 
// Testing frequency 132 Hz
08:23:04 
// Testing frequency 133 Hz
08:23:04 
!! {"code": "", "msg":"MCU 'mcu' shutdown: Move queue overflow", "values": []}
08:23:04 
!! {"code": "", "msg":"MCU 'mcu' shutdown: Move queue overflow", "values": []}
08:23:04 
!! {"code": "", "msg":"MCU 'mcu' shutdown: Move queue overflow", "values": []}

Guilouz, 2023-11-29:
This means your MCU is overloaded. Resonance tests require a lot of resources. Make sure the printer is not doing anything else at the same time and that its resources are not already occupied.

closed 2023-11-30 by Guilouz

shaidou1287, 2024-01-09:
I know this has been closed but I am currently having this exact issue. Nothing is being done on the printer... I restart Firmware and it is the first thing I run.

NeverUsedID, 2024-01-09:
you need to powercycle your printer, restart is not enough. At least it works this way on my side this way.