Guilouz/Creality-Helper-Script-Wiki #151: Klipper Git Repo Detached

momonator25, 2023-11-29:
### What happened?

I followed the install guide after unlocked root accsess with the newest firmware (1.3.2.20) from creality  with the helper script. Everything seems to work but the Klipper Repo is detached from Git.

### What did you expect to happen instead?

Update is possible via the _Update Manager_ as described [here](https://github.com/Guilouz/Creality-K1-and-K1-Max/wiki/Update-Manager)
[image]

### How to reproduce this issue?

- Install newest firmware (1.3.2.20)
- Install Mainsail via Helper Script
- Go to Mainsail Web Frontend
- Go to Machine

momonator25, 2023-11-29:
no if have not, should I?

Guilouz, 2023-11-29:
Yes… It’s indicated on the wiki.

momonator25, 2023-11-29:
Oh ok my bad. It's because I have installed the rooted firmware OTA not with this guide. I guess we can close this. Sorry and thanks for the fast reply!

closed 2023-11-29 by Guilouz