Guilouz/Creality-Helper-Script-Wiki #162: [BUG] CPU sensor type may be wrong in fans configuration

ex0rcist, 2023-12-02:
### What happened?

I copied your fan configuration and it shows different temperatures on the sensor and on the fan (and are also called differently, "mcu temp" vs "cpu temp"). I Have K1 max. My configuration: https://gist.github.com/ex0rcist/6cfa52712b9de6e79b63b08a97236e49

### What did you expect to happen instead?

It is expected that the temperatures on the sensor and on the fan will be the same

### How to reproduce this issue?

K1 max, 1.3.2.20, https://gist.github.com/ex0rcist/6cfa52712b9de6e79b63b08a97236e49

### Additional information:

I'm not the only one - https://github.com/Guilouz/Creality-K1-and-K1-Max/discussions/148

ex0rcist, 2023-12-02:
[image]
difference between them is 5-6 deg

Guilouz, 2023-12-02:
Already answered in other PR.