Guilouz/Creality-Helper-Script-Wiki #182: [BUG] Moonraker dependencies out-of-date causing exception when updating through update manager

caffeineflo, 2023-12-08:
### What happened?

I've updated moonraker through the built-in update manager to the latest commit. Since moonraker and the python virtualenv are prepackaged in this repo there was a mismatch on the dependencies that moonraker needs to fully run. 
When trying to connect the latest version of OrcaSlicer with Moonraker or either one of the frontends (Mainsail/Fluidd) I received this error:

```
Could not connect to OctoPrint: HTTP 404: {"error": {"code": 404, "message": "Not Found",
"traceback": "Traceback (most recent call last):\n\n File
\"/usr/data/moonraker/moonraker-env/lib/python3.8/site-packages/tornado/web.py\", line 1681, in
_execute\n result = self.prepare()\n\n File \"/usr/data/moonraker/moonraker/moonraker/app.py\", line
1036, in prepare\n raise tornado.web.HTTPError(404)\n\ntornado.web.HTTPError: HTTP 404: Not
Found\n"}}
```

I spent quite a bit of time during the last few days trying to both natively compile all the dependencies moonraker has on my K1 unsucessfully and now tried to both cross compile as well as using the buildroot I found elsewhere to build the python virtualenv up. Unfortunately I haven't been successful with that yet.

### What did you expect to happen instead?

OrcaSlicer -> Moonraker: Connection success

Moonraker Update -> Update dependencies or avoid the update

### How to reproduce this issue?

Update moonraker to the latest commit, then try to connect OrcaSlicer to Moonraker/Fluidd/Mainsail

### Additional information:

Would love to know how you put together the virtualenv in the moonraker.tar - I'm happy to update it and put a PR up if you tell me how that was compiled

Guilouz, 2023-12-09:
It's the same Moonraker provided by Creality and updated with git difference with command line.

I have latest Moonraker build without issue:

[image]

[image]

closed 2023-12-10 by Guilouz

caffeineflo, 2023-12-11:
FWIW, I found the issue and was able after some additional work to compile all the recent dependencies. I'll put up a PR tomorrow to update the files