Guilouz/Creality-Helper-Script-Wiki #189: [BUG] Can't reinstall Camera Settings Control

cinealfa, 2023-12-11:
### What happened?

Did an uninstall of the Camera Settings Control. when I tried to reinstall, nothing happens.

### What did you expect to happen instead?

Files to be reinstalled

### How to reproduce this issue?

It happens every time, I've rebooted the unit, restarted the script, updated the script, and the behavior doesn't change.

Guilouz, 2023-12-11:
All functions are tested and working.

What error do you have ?

cinealfa, 2023-12-12:
No error per se, When I run the install script there is no dialog asking if I want to install. Instead, the number disappears like it has done the install. When I run the script to list what is installed the camera setting are listed as not installed. In addition and maybe unrelated, the webcam works during printing but no timelapse is recorded.

cinealfa, 2023-12-18:
I figure out the problem : The Kilpper command line needs to be installed. Apparently, I did not do that first. It might be worthwhile to add a line of text in the description explaining what features are dependent on it. So boneheads like me don't screw up the install.

closed 2023-12-18 by cinealfa