Guilouz/Creality-Helper-Script-Wiki #205: [BUG] Moonraker is invalid

matayom, 2023-12-21:
### What happened?

Update manager in Fluidd says that moonraker is invalid, and I cannot update it. Not sure if its related but every 10 prints or so my K1 crashes randomly

### What did you expect to happen instead?

Successful update

### How to reproduce this issue?

No idea

### Additional information:

How do i manually update moonraker from terminal?

matayom, 2023-12-21:
If i try to click "Recover" i get an error: 
`Application moonraker: Attempting Repo Recovery...
Error Recovering moonraker`
and
`Repo has not been verified, clone aborted`

Guilouz, 2023-12-21:
Try to remove and reinstall it

matayom, 2023-12-21:
Can i remove and reinstall **only** moonraker with your helper script without messing anything up? Sorry this is my first time using klipper/moonraker/etc

hjoungjoo, 2023-12-21:
When updated to v1.3.2.20, there is a conflict with the existing installation file.
I had to delete it all with the installation tool and reinstall it from scratch.

matayom, 2023-12-21:
> When updated to v1.3.2.20, there is a conflict with the existing installation file. I had to delete it all with the installation tool and reinstall it from scratch.

Simply reinstalling fixed it or is the conflict persistent?

hjoungjoo, 2023-12-23:
> > When updated to v1.3.2.20, there is a conflict with the existing installation file. I had to delete it all with the installation tool and reinstall it from scratch.
> 
> Simply reinstalling fixed it or is the conflict persistent?

Simply reinstalling fixed.

jacquesglerner, 2023-12-26:
> > > When updated to v1.3.2.20, there is a conflict with the existing installation file. I had to delete it all with the installation tool and reinstall it from scratch.
> > 
> > 
> > Simply reinstalling fixed it or is the conflict persistent?
> 
> Simply reinstalling fixed.

I have done the reinstall but the particular version that is installed ends in 208 and the last one ends in 223. Not a big deal. Just pointing.

closed 2023-12-29 by Guilouz