Guilouz/Creality-Helper-Script-Wiki #207: [BUG] Mainsail hangs at Inititalizing screen.

The-Monkey-King, 2023-12-21:
### What happened?

After using the Helper script, Mainsail does not load. Hangs at "Initializing" and browser tab title says, "ERROR." Does not load on Bing, Chrome, and Firefox (my main browser).

Looking at Moonraker log, I see some **common.py:build_error** statements. 
Example:
`[common.py:build_error()] - JSON-RPC Request Error - Requested Method: server.connection.identify, Code: -32602, Message: Unauthorized`

Restarted at system level (moonraker and nginx): 3 times
Hardware restart: 2 times

### What did you expect to happen instead?

Expected Mainsail's main screen to load.

### How to reproduce this issue?

1. Updated successfully to Firmware 1.3.2.20
2. Established wifi access to Internet
3. Successfully loaded and ran helper script (as of dec 21 2023)
4. Installed in this order: (1)Moonraker and Nginx, (3)Mainsail, (4) Entware, (5)Klipper GCode Shell Command, (7)Fix for Reboot..., (12) Camera Settings Control
5. Opened browser (Firefox 120.0.1 64-bit) to https://<K1 LAN Adrees>:4409
6. Initializing screen with moving bar stays. Does not change to main screen. Tab title is, "ERROR."
7. Made sure no antivirus add-ins were active.
8. Tested on Bing and Chrome with same result.

### Additional information:

[moonraker.log](https://github.com/Guilouz/Creality-K1-and-K1-Max/files/13744444/moonraker.log)
[klippy.log](https://github.com/Guilouz/Creality-K1-and-K1-Max/files/13744446/klippy.log)

Attached Moonraker and klipper logs.

The-Monkey-King, 2023-12-21:
UPDATE:
**The giveaway here is the error, server connection unauthorized.** The LAN IP address of the printer was not carried over into the moonraker configuration file during installation of Moonraker. I know, I know. Most of these first time 3d printer users are most likely not educated in network administration, hence the preloaded 192... default local host network. But this will always be an issue with a user defined network. 

For file: 
**/usr/data/printer_data/config/moonraker.conf**

In section:
**[authorization]
trusted clients:**

Add:
**X.Y.0.0/16**          #where X and Y are the first two octets of the user's LAN, IPv4

This allows the whole LAN to be trusted and makes allowances for the printer to change IP addresses due to reboots, restarts, etc. of the network. No hard coding of the current printer's IP address is necessary. In fact, it would be discouraged.

HOW TO?
- Method 1: Query IP address, mask the last two octets, and then add to trusted clients as a string? (like "IpAddress2String")
- Method 2: Use Curl to convert it text and add it as a text string
- Method 3: Use sed on reboots to change the IP address

The first two occur during the installation of Moonraker and the last one after each printer reboot.

Guilouz, 2023-12-22:
My moonraker.conf already have trusted clients:

trusted_clients:
  10.0.0.0/8
  127.0.0.0/8
  169.254.0.0/16
  172.16.0.0/12
  192.168.0.0/16
  FE80::/10
  ::1/128

The-Monkey-King, 2023-12-23:
Yes, I see you put in the generic IP ranges to capture most home networks. This error only occurs for users with a network that is not generic.

Can you keep this open for 2-3 weeks? I think I have a solution but this being Christmas and New Years in the US for the next 10 days, I may not have it readily available right now.

closed 2024-01-07 by Guilouz