Guilouz/Creality-Helper-Script-Wiki #214: [BUG] Camera Settings won't install

centauri61032, 2023-12-26:
### What happened?

Using v3.9.1 of the installation script. When pressing '12' from the Install Menu to install Camera Settings Control Files, nothing happens. 

### What did you expect to happen instead?

Install the settings.

### How to reproduce this issue?

Try to install item 12 from Install Menu

Guilouz, 2023-12-26:
I can't reproduce issue. Camera settings install correctly.

NeverUsedID, 2023-12-27:
i always have the problem that the date from my k1 max is completly wrong on console (correct on fluidd), so ssl certifiactes are always invalid. Had this on Camera install too, if i remember correctly. If i change the date with 'date --set="2023-12-27 12:43"' installation works.
Is there a way to autoupdate the date on reboot?

cinealfa, 2023-12-28:
Did you install Gcode shell first, The camera settings wouldn't install before that for me.

centauri61032, 2023-12-28:
> 

This worked.
Thanks!!

centauri61032, 2023-12-28:
> Did you install Gcode shell first, The camera settings wouldn't install before that for me.

This worked.
Thanks!

closed 2023-12-29 by Guilouz