Guilouz/Creality-Helper-Script-Wiki #216: Moonraker hat fehlerhafte Plugins

alenba90, 2023-12-27:
### What happened?

After the Moonraker update, the following error occurs when booting
[image]

### What did you expect to happen instead?

after the update everything should work normally

### How to reproduce this issue?

after reinstalling Moonraker ( I also re-rooted K1 and reinstalled Klipper)  , the same error occurs.

### Additional information:

Moonraker has failed plugins, please check your logs, update your configuration and restart moonraker.
history
Moonraker warnings found.
Component 'history' failed to load with error: 'Cursor' object has no attribute 'getmulti'
Moonraker plugin configuration can be found

jrezin, 2023-12-27:
Same thing happened after trying to update moonraker in the software update. It seems that the problem, at least here, lies with the "pillow==10.0.1" pip package.
maybe it's trying to mess with something outside /usr/data...

Rushent08, 2023-12-28:
Having the same problem. Only just rooted my K1 Max only to be greeted by this. 
[moonraker (3).log](https://github.com/Guilouz/Creality-K1-and-K1-Max/files/13782254/moonraker.3.log)

Guilouz, 2023-12-28:
It's Moonraker issue since recent PR.

closed 2023-12-29 by alenba90