Guilouz/Creality-Helper-Script-Wiki #219: Input Shaper not working after update[BUG] <title>

Myklt8, 2023-12-28:
### What happened?

Updated to 1.32.31 and updated and reinstalled all modules, however, the input shaper gives an error and shuts the machine down. It also reverts the cfg file to factory settings

### What did you expect to happen instead?

Input shaper to work as it did before

### How to reproduce this issue?

[klippy (2).log](https://github.com/Guilouz/Creality-K1-and-K1-Max/files/13788104/klippy.2.log)
[image]

alenba90, 2023-12-28:
[image]

alenba90, 2023-12-28:
missing in [gcode_macro Qmode] following cod:

`variable_max_accel: 2500
variable_max_accel_to_decel: 2500`

INPUT_SHAPER works after inserting code

Guilouz, 2023-12-28:
@alenba90 after fw update you must restore default settings to have latest config files.
There is some changes in latest firmware

Myklt8, 2023-12-29:
I did a factory reset and reloaded everything, the error went away, but the machine shuts off when doing the inputshaping...when it switches to the Y axis. mcu rpi shut down and rescheduled timer in the past

iosub, 2023-12-30:
Hi, is  1.32.31 firmware compatible with K1 or K1-max?

Myklt8, 2023-12-31:
Yes, it supports both

VinoY27, 2024-01-01:
I too faced the same issue, as said, Update the firmware, reset to factory settings, then follow the steps as per initial setup, root it and then install all the required things as per the Wiki.

closed 2024-01-03 by Guilouz