Guilouz/Creality-Helper-Script-Wiki #221: [BUG] <title>

pwilliams105, 2023-12-29:
### What happened?

Hi,
When i am trying to connect to my printer through fluidd I am getting the below error.
I have tried uninstalling fluid moonraker, etc but still no luck

Moonraker warnings found.
Unable to find DBus PolKit Interface, this suggests PolKit is not installed on your OS.

Any help would be greatly appreciated

### What did you expect to happen instead?

Connection to the printer would open normally to wirelessly send a print

### How to reproduce this issue?

If i close down my fluidd tab on chrome and re open it it displays the same issue
[image]

Guilouz, 2023-12-29:
Check your moonraker.conf file.

pwilliams105, 2023-12-30:
The man him self!!! Firstly, Thank you for all the hard work you have done with the k1 and k1 Max it is much appreciated.
And secondly, what would I need to check within the moonraker config file, I have tried chat gpt, google with no prevail on what to alter.
Thank you in advance

Guilouz, 2023-12-30:
Post your mooonraker.conf file

pwilliams105, 2023-12-30:
Thanks again, i am very new to all this but very keen to learn.

[server]
host: 0.0.0.0
port: 7125
klippy_uds_address: /tmp/klippy_uds
max_upload_size: 1024

[file_manager]
queue_gcode_uploads: False
# Set to 'True' if you use KAMP
enable_object_processing: False

[database]

[data_store]
temperature_store_size: 600
gcode_store_size: 1000

[machine]
provider: none
validate_service: False
validate_config: False

[authorization]
force_logins: False
cors_domains:
  *.local
  *.lan
  *://app.fluidd.xyz

trusted_clients:
  10.0.0.0/8
  127.0.0.0/8
  169.254.0.0/16
  172.16.0.0/12
  192.168.0.0/16
  FE80::/10
  ::1/128

[octoprint_compat]

[history]

[update_manager]
enable_auto_refresh: True
refresh_interval: 24
enable_system_updates: False

# Remove '#' from the lines below to enable camera configuration with Moonraker and replace 'xxx.xxx.xxx.xxx' by your IP addresses
#[webcam Camera]
#location: printer
#enabled: True
#service: mjpegstreamer
#target_fps: 15
#target_fps_idle: 5
#stream_url: http://xxx.xxx.xxx.xxx:8080/?action=stream
#snapshot_url: http://xxx.xxx.xxx.xxx:8080/?action=snapshot
#flip_horizontal: False
#flip_vertical: False
#rotation: 0
#aspect_ratio: 4:3

# Remove '#' from the lines below if you use Timelapse function and replace port '4408' by '4409' in snapshoturl if you use Mainsail
#[timelapse]
#output_path: /usr/data/printer_data/timelapse/
#frame_path: /usr/data/printer_data/frames/
#snapshoturl: http://localhost:4408/webcam/?action=snapshot

# Remove '#' from the lines below if you use Fluidd
#[update_manager fluidd]
#type: web
#channel: beta
#repo: fluidd-core/fluidd
#path: /usr/data/fluidd

# Remove '#' from the lines below if you use Mainsail
#[update_manager mainsail]
#type: web
#channel: beta
#repo: mainsail-crew/mainsail
#path: /usr/data/mainsail

# Remove '#' from the lines below if you use KAMP
#[update_manager KAMP-for-K1-Series]
#type: git_repo
#channel: dev
#path: /usr/data/KAMP-for-K1-Series
#origin: https://github.com/Guilouz/KAMP-for-K1-Series.git
#primary_branch: main
#is_system_service: False

Guilouz, 2023-12-30:
Post the moonraker.conf file directly

pwilliams105, 2023-12-30:
[moonraker.conf_1703933435562.zip](https://github.com/Guilouz/Creality-K1-and-K1-Max/files/13797805/moonraker.conf_1703933435562.zip)

pwilliams105, 2023-12-30:
Is that better?

Guilouz, 2023-12-30:
Your file is ok. Can you post the entire error.

pwilliams105, 2023-12-30:
[image]

Guilouz, 2023-12-30:
I have no idea why you have this warning, provider: none is already set to none

[machine]
provider: none

See: https://github.com/Arksine/moonraker/issues/349

pwilliams105, 2023-12-30:
Is there a way to completely reset the motherboard to factory settings and firmware? Like it has just been sent from factory?

Guilouz, 2023-12-30:
https://github.com/Guilouz/Creality-K1-and-K1-Max/wiki/Reset-Factory-Settings

pwilliams105, 2024-01-01:
Thanks mate, i appreciate your help.
Unfortunately I ran into the same issue after a few tries, resetting and following your script.
I ran the fluid installation manually and then added your camp installation, unfortunately when I added the auto-update feature for KAMP and fluid I began to get connection errors and the following error. 
DO you have any ideas why this would be?

pwilliams105, 2024-01-01:
[image]

Guilouz, 2024-01-03:
You must have this in your moonraker.conf file

[update_manager]
enable_auto_refresh: True
refresh_interval: 24
enable_system_updates: False

closed 2024-01-07 by Guilouz