Guilouz/Creality-Helper-Script-Wiki #222: [BUG] How to change the position of the purge line?

gfBarreto, 2023-12-29:
### What happened?

Purge line being printed on the part.

### What did you expect to happen instead?

I'm using the K1.
Sometimes the purge line remains part of the part.
I traditionally use it on the side on the S1 Pro.
How do I change the location?
Attached I send a piece already printed and another that started printing and I had to stop.
[image]
[image]

### How to reproduce this issue?

.

Guilouz, 2023-12-29:
Not possible without to modify macros code.

Please post issue only for issue. That’s is not an issue.

closed 2023-12-29 by Guilouz

gfBarreto, 2023-12-29:
ok, obrigado.

Tombraider2006, 2024-01-01:
this error occurs when there is no object exception in the slicer. set the flag and there will be no such error.