Guilouz/Creality-Helper-Script-Wiki #223: [BUG] <cannot download fluidd or mainsail>

thegreat7man, 2023-12-29:
### What happened?

when i try to install fluidd or mainsail im getting `Downloading Fluidd file...
Connecting to github.com (140.82.121.3:443)
wget: TLS error from peer (alert code 80): 80
wget: error getting response: Connection reset by peer
 Download failed. Exit code: 1`

### What did you expect to happen instead?

fluidd or mainsail would be installed

### How to reproduce this issue?

go to the help script, install, 2 (fluidd)

Guilouz, 2023-12-29:
Read the wiki…

closed 2023-12-29 by Guilouz

thegreat7man, 2023-12-29:
and solve of the problem with entware is not working

thegreat7man, 2023-12-29:
```
root@K1-EDE6 /root [#] opkg install wget-ssl
-sh: opkg: not found
```

thegreat7man, 2023-12-29:
do i need to open new issue ?

Guilouz, 2023-12-29:
That’s means tou have not installed entware

thegreat7man, 2023-12-29:
i have installed ` Installing Entware...

Making /opt directory on data partition where there is space, and adding a symbolic link...
Installing Entware using generic installer script...
Connecting to bin.entware.net (158.160.45.54:80)
writing to stdout
Info: Checking for prerequisites and creating folders...
Warning: Folder /opt exists!
-                    100% |********************************|  2760  0:00:00 ETA
written to stdout
Info: Opkg package manager deployment...
Connecting to bin.entware.net (158.160.45.54:80)
wget: can't connect to remote host (158.160.45.54): Connection timed out
chmod: /opt/bin/opkg: No such file or directory
Connecting to bin.entware.net (158.160.45.54:80)
saving to '/opt/etc/opkg.conf'
opkg.conf            100% |********************************|   150  0:00:00 ETA
'/opt/etc/opkg.conf' saved
Info: Basic packages installation...
/bin/sh: /opt/bin/opkg: not found
/bin/sh: /opt/bin/opkg: not found
Info: Congratulations!
Info: If there are no errors above then Entware was successfully initialized.
Info: Add /opt/bin & /opt/sbin to $PATH variable
Info: Add "/opt/etc/init.d/rc.unslung start" to startup script for Entware services to start
Info: Found a Bug? Please report at https://github.com/Entware/Entware/issues
Adding /opt/bin and /opt/sbin to the start of the PATH in the system profile...
Adding startup script...

 Entware has been installed successfully!

 Log out and log back in, and you can now install packages with: opkg install <packagename>`

thegreat7man, 2023-12-29:
but its going on with this issue

thegreat7man, 2023-12-29:
maybe there an option to install it from usb localy ? or do i need to change wifi ?

thegreat7man, 2023-12-29:
oh, and i have updated firmware to 1.3.2.31

crackers81, 2023-12-29:
Same problem here.

thegreat7man, 2023-12-29:
please,help me somebody because i cant use printer without it

crackers81, 2023-12-29:
> please,help me somebody because i cant use printer without it

DO THIS after installing entware
(When it's done, log out of the current SSH session and log in again.)

ONealP, 2023-12-29:
I keep having the same issue. I log out of current SSH and back in. Same issue. I have reset printer to factory. re-rooted and now cant even install helper installer.

sh: /usr/bin/xauth: not found
root@K1Max-6B13 /root [#] cd && wget --no-check-certificate https://raw.githubusercontent.com/Guilouz/Creality-K1-a
nd-K1-Max/main/Scripts/installer.sh
Connecting to raw.githubusercontent.com (185.199.108.133:443)
wget: got bad TLS record (len:0) while expecting switch to encrypted traffic
wget: error getting response: Connection reset by peer

Guilouz, 2023-12-29:
That's mean your printer have not the correct date, type this command:

date

and see result

ONealP, 2023-12-30:
Sorry right after I posted I got called in to work.
typed 
Date and it gave me the correct date and time.

root@K1Max-6B13 /root [#] date
Fri Dec 29 20:50:09 CST 2023

thegreat7man, 2023-12-30:
im getting this when installing entware `wget: can't connect to remote host (158.160.45.54): Connection timed out`

thegreat7man, 2023-12-30:
@Guilouz i guess there is the problem even with entware installation

Guilouz, 2023-12-30:
No entware installation working fine, issue is on your machine or network.

thegreat7man, 2023-12-30:
@Guilouz i tried few wifi sources and nothing. maybe a new firwmare is bad and i should do downgrade ?

thegreat7man, 2023-12-30:
this ip is dead

Guilouz, 2023-12-30:
I use latest fw without issue on my max

thegreat7man, 2023-12-30:
so what sould i do now? i tried any devices,many wifi's, vpn and nothing. please, help

thegreat7man, 2023-12-30:
@Guilouz i guess i got the problem. entware's server (158.160.45.54:80) is located in russia and i live in ukraine with are blocking all russian servers. do i have some options to install it in alternative way ?

thegreat7man, 2023-12-30:
maybe you can make any mirror for it or something else ?

Guilouz, 2023-12-30:
Have you access to this URL ? http://bin.entware.net/mipselsf-k3.4/installer/

thegreat7man, 2023-12-30:
yep

thegreat7man, 2023-12-30:
but cannot with clean ip adress (158.160.45.54:80)

thegreat7man, 2023-12-30:
maybe there are any mirrors exist ?

Guilouz, 2023-12-30:
it's the url used in my script and it's the official website.

thegreat7man, 2023-12-30:
hm... but why it cant connect. previous time i didnt install entware because fluidd installed without it. so maybe i can somehow install it another way ?

thegreat7man, 2023-12-30:
can i try to use something like mobaxterm and manually install entware ? but for this i need to know wich file and where to place

Guilouz, 2023-12-30:
You can find the install process here:

https://github.com/Guilouz/Creality-K1-and-K1-Max/blob/054da79a7f9d12a9e3f76bd90607b53dcac401eb/Scripts/installer.sh#L632