Guilouz/Creality-Helper-Script-Wiki #224: [BUG] <entware is not installing in Ukraine>

thegreat7man, 2023-12-30:
### What happened?

the script cant get access to the entware server because Ukraine has been blocked all Russian servers

### What did you expect to happen instead?

entware could have install

### How to reproduce this issue?

try to install entware from the script in Ukraine

Guilouz, 2023-12-30:
Please stop open many issue for the same problem.

closed 2023-12-30 by Guilouz

thegreat7man, 2023-12-30:
but its the problem and you dont want to even try to help

thegreat7man, 2023-12-30:
@Guilouz you are ignoring me but at least you can try to help me to somehow solve this. maybe i can somehow install entware manually ?

Guilouz, 2023-12-30:
I have a life, I'm not always there in front of my github...
Issue is for reporting a github related issue and not for cross country issues...
You've already created an issue for this, so there's no need to create another one!

thegreat7man, 2023-12-30:
but im not guilty of where i living. i just want to install your awesome script (

kisslorand, 2024-01-06:
One can always use a VPN...

Guilouz, 2024-01-06:
I have fixed this with the new script version.

thegreat7man, 2024-01-06:
oh,thanks a lot ! have a nice day