Guilouz/Creality-Helper-Script-Wiki #230: Fluidd and Mainsail cannot be installed

user37337, 2023-12-31:
### What happened?

First of all: I appreciate your work. Thank you!
Now my issue: I've downloaded the installer and installing of moonraker works fine. Then I tried to install Fluidd and Mainsail but both exits with the same error message:

```
 Installing Fluidd...

Downloading Fluidd file...
Connecting to github.com (140.82.121.4:443)
wget: TLS error from peer (alert code 80): 80
wget: error getting response: Connection reset by peer
 Download failed. Exit code: 1
```

Is there anything I've done wrong?

### What did you expect to happen instead?

Installing should be successful

### How to reproduce this issue?

-

Guilouz, 2023-12-31:
Read the wiki…

user37337, 2023-12-31:
Sorry, didn't knew the wiki exists. Works fine now. Thank you!

closed 2023-12-31 by user37337