Guilouz/Creality-Helper-Script-Wiki #231: Creality print GCodeMeta thumbnails do not appear in fluidd

User09342759075-93275-23957, 2024-01-01:
### What happened?

I believe that there is no data from gcode after the Moonraker update.
According to the history of the press, I saw that it was from December 21st

### What did you expect to happen instead?

Get gcode Meta data and thumbnails from /tmp/creality/local_gcode/gmail folder/

### How to reproduce this issue?

If there is no problem in the transfer from Creality Print, then I am the only one with the problem.
[image]

Guilouz, 2024-01-01:
Moonraker have not the Creality metadata

User09342759075-93275-23957, 2024-01-01:
Is it possible to fix it as it was before?

Guilouz, 2024-01-01:
To do this, Moonraker would need to integrate Creality Print metadata.

User09342759075-93275-23957, 2024-01-02:
I have found a solution https://github.com/ballaswag/moonraker/blob/61ae2cc8c8e4f41a7bc082591d4d06fabe4e8065/moonraker/components/file_manager/metadata.py
[image]
replaced the file metadata.py not all the data, but there is a result.

cinealfa, 2024-01-03:
Good find, I did some research why it wasn't pulled into the codebase Here is a quote from the developer of moonraker:
"Unfortunately Creality continues to claim that the Creality Print Slicer is their own "self-developed" software, when it is clear from the metadata that it is a fork of Cura. They do not give credit to Ultimaker nor are they in compliance with the GPL. This is in addition to initially claiming that the K1 firmware was their own creation. It wasn't until the device was rooted and dumped did they admit that they were using modified versions Klipper, Moonraker, Fluidd.

As such, I will not merge specific support for Creality hardware or software in this repo." 
We need to remind Creality that we are watching them...

Guilouz, 2024-01-04:
Yes but modifying this file will result in a dirty repo and breaks the update manager:

[image]

closed 2024-01-05 by User09342759075-93275-23957