Guilouz/Creality-Helper-Script-Wiki #236: [BUG] KAMP Line Purge

Vitorleoron, 2024-01-02:
### What happened?

When the STL file has brim enable, the width of the brim is not consider to calculate the object size. So the KAMP purge line is inside the brim region!

### What did you expect to happen instead?

The purge line should be after the brim and the skirt. This wasn't my case, but if the purge line exceds the bed limits, I think that change the region to Y or in the region where are made the nozzle cleaning is a good solution!

### How to reproduce this issue?

Print an object with brim and skirt enable using KAMP!

The purge line will be between the skirt and the object, in the brim perimiter!

mindkill666, 2024-01-02:
I have this too sometimes, i changed the line in kamp_settings.cfg to:  variable_purge_margin: 25
I gives  you more space between the purge line and the brim.

Vitorleoron, 2024-01-02:
Thanks, I will make this change!

closed 2024-01-10 by Guilouz