Guilouz/Creality-Helper-Script-Wiki #238: [BUG]BED_LEVELING

alenba90, 2024-01-02:
### What happened?

Sometimes a strange bed leveling happens.
 When he does the bed level calibration once, he repeats it again and prepares the measurements twice, so that in the end I get a very bad bed.

[image]

When the adaptive bed calibration works, then I have no problems, This problem only occurs when I manually calibrate the entire bed.

**I also get an error in the console,when I do manual calibration:** _No objects detected! Check your gcode and make sure that EXCLUDE_OBJECT_DEFINE is happening before BED_MESH_CALIBRATE is called. Defaulting to regular meshing._

I rooted the machine several times. It's strange that I managed to make a good Bed Mesh at the beginning. after that, the game is a matter of luck when they will make a good Calibration.

### What did you expect to happen instead?

I expect to be able to periodically calibrate the entire bed, so I could turn off calibration for each print and use the default bed calibration for that

### How to reproduce this issue?

when I set the temperature of the bed to some value, for example 80 degrees during calibration, then it happens that it calibrates the bed 2 times.
Or if I don't delete the old Mesh. I noticed that the old Mesh sometimes creates a problem when recalibrating
if I don't set a temperature for the bed, then it calibrates the entire base normally

### Additional information:

[config-20240102-204420.zip](https://github.com/Guilouz/Creality-K1-and-K1-Max/files/13813400/config-20240102-204420.zip)

alenba90, 2024-01-03:
when i change on [bed_mesh] probe_count: from 6,6 to 5,5. then everything works normally.
It seems that there is some problem when doing mesch with 6,6

[bed_mesh]
speed: 150
mesh_min: 5,5
mesh_max: 295,295
probe_count: 5,5
fade_start: 5.0
fade_end: 50.0
algorithm: bicubic
bicubic_tension: 0.1

Guilouz, 2024-01-03:
Nothing to see with the wiki here.
Please only post issue related to the wiki.

closed 2024-01-03 by Guilouz