Guilouz/Creality-Helper-Script-Wiki #248: Source contains parsing errors: '/usr/data/printer_data/config/gcode_macro.cfg'

Jamespev, 2024-01-06:
### What happened?

Ive followed the process throught but after the restart im getting the following error on fluidd:

Source contains parsing errors: '/usr/data/printer_data/config/gcode_macro.cfg'
[line 288]: 'SMART_PARK\n'
[line 289]: 'M109 S{extruder_temp}\n'
[line 290]: 'M190 S{bed_temp}\n'
[line 291]: 'LINE_PURGE\n'

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

### What did you expect to happen instead?

Not get any errors

### How to reproduce this issue?

Ive followed the tutorial twice with the same issues

### Additional information:

# K1-MAX
# CreateDate: 2023/03/21

[gcode_macro PRINTER_PARAM]
variable_z_safe_pause: 0.0
variable_z_safe_g28: 3.0
variable_max_x_position: 300.0
variable_max_y_position: 300.0
variable_max_z_position: 300.0
variable_fans: 3
variable_auto_g29: 0
variable_fan0_min: 25
variable_fan1_min: 50
variable_fan2_min: 180
variable_fan2_speed: 0
variable_hotend_temp: 0
variable_e_min_current: 0.27
gcode:

[gcode_macro AUTOTUNE_SHAPERS]
variable_autotune_shapers: 'ei'
gcode:

[gcode_macro LOAD_MATERIAL_CLOSE_FAN2]
variable_fan2_value: 0
gcode:
  SAVE_GCODE_STATE NAME=myMoveState
  {% if printer['output_pin fan2'].value > 0.0 %}
    SET_GCODE_VARIABLE MACRO=LOAD_MATERIAL_CLOSE_FAN2 VARIABLE=fan2_value VALUE={printer['output_pin fan2'].value}
    M107 P2
  {% endif %}
  RESTORE_E_CURRENT

[gcode_macro LOAD_MATERIAL_RESTORE_FAN2]
gcode:
  {% set fan2_value = printer['gcode_macro LOAD_MATERIAL_CLOSE_FAN2'].fan2_value|float %}
  RESTORE_GCODE_STATE NAME=myMoveState
  {% if fan2_value > 0.0 %}
    {% set s_value = (fan2_value * 255 - printer["gcode_macro PRINTER_PARAM"].fan2_min) * 255 / (255 - printer["gcode_macro PRINTER_PARAM"].fan2_min) %}
    M106 P2 S{s_value}
    SET_GCODE_VARIABLE MACRO=LOAD_MATERIAL_CLOSE_FAN2 VARIABLE=fan2_value VALUE=0
  {% endif %}
  SET_E_MIN_CURRENT

[gcode_macro SET_E_MIN_CURRENT]
gcode:
[... 570 lines cut ...]
#     BED_MESH_OUTPUT

[gcode_macro DEFINE_OBJECT]
gcode:
  EXCLUDE_OBJECT_DEFINE {rawparams}

[gcode_macro START_CURRENT_OBJECT]
gcode:
  EXCLUDE_OBJECT_START NAME={params.NAME}

[gcode_macro END_CURRENT_OBJECT]
gcode:
  EXCLUDE_OBJECT_END {% if params.NAME %}NAME={params.NAME}{% endif %}

[gcode_macro LIST_OBJECTS]
gcode:
  EXCLUDE_OBJECT_DEFINE

[gcode_macro LIST_EXCLUDED_OBJECTS]
gcode:
  EXCLUDE_OBJECT

[gcode_macro REMOVE_ALL_EXCLUDED]
gcode:
  EXCLUDE_OBJECT RESET=1

Guilouz, 2024-01-06:
What process did you follow? This lacks information.

And send the file rather than copying and pasting the content that github reformats.

Jamespev, 2024-01-06:
I followed the links below 

https://github.com/Guilouz/Creality-K1-and-K1-Max/wiki/Installation-Helper-Script

https://github.com/Guilouz/Creality-K1-and-K1-Max/wiki/Use-KAMP

I tried to attatch the file but it wouldnt let me.

Guilouz, 2024-01-06:
This is not needed anymore, you have just to install KAMP with latest script version.

closed 2024-01-06 by Jamespev