Guilouz/Creality-Helper-Script-Wiki #251: [BUG] Failures installing on 1.3.3.3 firmware with 4.0 installer.

jgefaell, 2024-01-06:
### What happened?

Got up this morning overjoyed to see the new firmware. Installed on my Max. Experienced these issues when installing modules. Any idea? The scripts are not in the init.d directory, though If found them elsewhere. Did a link command fail? Thanks, it's likely something I did out of order or a step I missed?

THANK YOU! 

When installing KAMP:
---
Restarting services...
no need to update custom config files
old_version: 1029
new_version: 1029
./installer.sh: line 1055: /etc/init.d/S56moonraker_service: not found
./installer.sh: line 1057: /etc/init.d/S56moonraker_service: not found
---

When installing Obico
---
 Obico is a Moonraker plugin that allows you to monitor
./installer.sh: line 1782: printf and control your 3D printer from anywhere.: not found

 Are you sure you want to install Obico ? (y/n): 
---

Restarting Moonraker and Nginx services
---

 Do you want to restart Moonraker and Nginx services ? (y/n): y

Restarting services...
./installer.sh: line 3492: /etc/init.d/S50nginx: not found
./installer.sh: line 3493: /etc/init.d/S56moonraker_service: not found
./installer.sh: line 3494: /etc/init.d/S50nginx: not found
./installer.sh: line 3496: /etc/init.d/S56moonraker_service: not found

 Moonraker and Nginx services have been restarted!
---

### What did you expect to happen instead?

Successful completion. 

### How to reproduce this issue?

I'm going to try that later today.

Guilouz, 2024-01-06:
Not a script issue.

It’s seems you have not moonraker and nginx installed

Have you reset to factory after updating fw?

jgefaell, 2024-01-07:
PEBCAK

closed 2024-01-07 by jgefaell