Guilouz/Creality-Helper-Script-Wiki #252: Input shaper calibration

lvlacleod, 2024-01-06:
### What happened?

After it runs calibration for y axis it generates 2 csv files. calibration_data_x_20240106_204244.csv and calibration_data_y_20240106_204244.csv. Both files seem to have the same data. This is before it begins the calibration for x axis which generates 1 file calibration_data_x_20240106_204715.csv

### What did you expect to happen instead?

I thought it did x axis calibration 1st of all and not y axis. I did not expect it to produce a csv file for x axis and y axis after it completed just the y axis calibration

### How to reproduce this issue?

I ran Input_shaper_calibration. Not sure if I am the only one experience this problem

### Additional information:

Installed latest firmware and latest script

Guilouz, 2024-01-06:
INPUT_SHAPER_CALIBRATION macro run Y axis first and then X axis to bypass overwritted X value.

Guilouz, 2024-01-07:
I have change the macro to run SHAPER_CALIBRATE command.

closed 2024-01-07 by Guilouz