Guilouz/Creality-Helper-Script-Wiki #257: [BUG] KAMP doesn't work when printing with Calibration from K1Max screen or Creality Print

lashley514, 2024-01-07:
### What happened?

When printing without calibration KAMP works from Orca or K1 Screen. When I check the calibration box on the screen or print with Calibration from Creality Print KAMP doesn't work and it meshes entire bed.

### What did you expect to happen instead?

KAMP to work in any situation

### How to reproduce this issue?

Print with Calibration

Guilouz, 2024-01-07:
Wich firmware ?

alenba90, 2024-01-07:
it's the same for me, on 1.3.3.3 firmware

lashley514, 2024-01-09:
> Wich firmware ?

1.3.3.3

dcs3473, 2024-01-09:
I am experiencing the same behavior.

weston-wright, 2024-01-09:
I posted in the facebook group but adding my observations here.
1.3.3.3 firmware

I sliced a simple xyz cube in both Creality Print and Orca Slicer.

If I print both of them through Fluidd, they both follow KAMP fine with 1 exception. The Creality Print file, after doing the purge, moves to the front left of the machine and then starts performing the print. I'm not sure why that would be as they both have the same start GCODE.

If I print both of them with Calibration (via the Creality Print portal or just off of the interface at the printer), BOTH sliced files do indeed go through all of the LIDAR calibration. But they do not do the KAMP bed leveling, it instead levels the entire bed. It does do the purge though and again the Creality Print sliced file does the same as above (after purge, moves to the 0,0 position and then goes to print).

Guilouz, 2024-01-09:
Yes there is some changes in 1.3.3.3 in screen UI side. I think it's related to the possibility we can run lidar test with other slicer and implementation of cancel object.

cinealfa, 2024-01-20:
it did not work with the previous version of the firmware either

SaturnsVoid, 2024-01-24:
I am thinking that when you use the screen to start a print its injecting a macro / gcode, and when you print from Mainsail/Fluidd/Orca its not getting that macro/gcode to run it... Probably have to figure out what it is and call it in the slicers start script?

3DP-Chch, 2024-01-26:
I have updated to 1.3.3.5 and when printing from screen with calibration it does a full bed mesh and without calibration it does no mesh. Printing through Fluidd works as expected.

Guilouz, 2024-02-03:
It's a limitation in latest firmware, can't do nothing.

closed 2024-02-03 by Guilouz