Guilouz/Creality-Helper-Script-Wiki #259: Improved Shapers Calibrations

PhilippeGalid, 2024-01-08:
### What happened?

Sorry but Install Point 12 Improved Shapers Calibrations Installation doesn't work, when I enter 12 nothing happens, I tried it on two computers, Mac and Windows on the same network, all other installations work. What kind of solution is there? They say there are no errors in the script, so I don't understand why the question doesn't come up whether I want to install it. You are doing a great job, please help, what can I do, Entware is also installed?

### What did you expect to happen instead?

Sorry but Install Point 12 Improved Shapers Calibrations Installation doesn't work, when I enter 12 nothing happens, I tried it on two computers, Mac and Windows on the same network, all other installations work. What kind of solution is there? They say there are no errors in the script, so I don't understand why the question doesn't come up whether I want to install it. You are doing a great job, please help, what can I do, Entware is also installed?

### How to reproduce this issue?

Sorry but Install Point 12 Improved Shapers Calibrations Installation doesn't work, when I enter 12 nothing happens, I tried it on two computers, Mac and Windows on the same network, all other installations work. What kind of solution is there? They say there are no errors in the script, so I don't understand why the question doesn't come up whether I want to install it. You are doing a great job, please help, what can I do, Entware is also installed?

### Additional information:

Sorry but Install Point 12 Improved Shapers Calibrations Installation doesn't work, when I enter 12 nothing happens, I tried it on two computers, Mac and Windows on the same network, all other installations work. What kind of solution is there? They say there are no errors in the script, so I don't understand why the question doesn't come up whether I want to install it. You are doing a great job, please help, what can I do, Entware is also installed?

Guilouz, 2024-01-08:
As I can said, it's working fine, see:

[image]

NeverUsedID, 2024-01-08:
check with "date" if your time is correct.

PhilippeGalid, 2024-01-08:
Hi thanks for your support i install 5 and after 12 works, thanks for your Support :-)

PhilippeGalid, 2024-01-08:
Works Now

closed 2024-01-08 by PhilippeGalid