Guilouz/Creality-Helper-Script-Wiki #261: [BUG] Removing OBICO leads to a ton of errors, lines are not being removed from Config files

riktanius, 2024-01-09:
### What happened?

I switched from using OBICO to Octoeverywhere and it seems that when I used the Remove Obico function it did remove all the Obico specific files, but the config files and update manager are now trying to call those files that dont exist anymore. They throw an error. 

### What did you expect to happen instead?

No issues when removing Obico

### How to reproduce this issue?

Install Obico, use it, the use the script to remove Obico

### Additional information:

[image]

There was also a major error, but I cleared that by commenting out Obico on any config files I found:

[image]

[image]

Guilouz, 2024-01-09:
Obico install and remove is not mine, it's from @kennethjiang.

Can you tell me witch settings are not removed exactly ?

kennethjiang, 2024-01-09:
@Guilouz Thank you for pinging me for this issue! Didn't realize creality would check `printer.cfg` by line number...

I was originally thinking about putting this line at the bottom of the `printer.cfg`. But Klipper bed level will generate this part at the bottom:

```
#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
...
```

Any suggestion on the best way to handle this?

Guilouz, 2024-01-09:
> @Guilouz Thank you for pinging me for this issue! Didn't realize creality would check `printer.cfg` by line number...
>
> I was originally thinking about putting this line at the bottom of the [...]

For my new script version I add includes after line:

[include printer_params.cfg]

An exemple here:

sed -i '/\[include printer_params\.cfg\]/a \[include Helper-Script/buzzer-support\.cfg\]' /usr/data/printer_data/config/printer.cfg

kennethjiang, 2024-01-09:
Gotcha! Thank you so much @Guilouz !

riktanius, 2024-01-10:
According to the Above, I should be able to remove the line from the "printer.cfg" file and everything ok?

Because I tried that and still getting this error:

[image]

riktanius, 2024-01-10:
Seems there was a line in Moonraker.conf that I had commented out but needed to just be deleted. Removed the line, restarted, error gone.

Guilouz, 2024-01-10:
@kennethjiang Did you fix it ?

kennethjiang, 2024-01-10:
Sorry have been really busy. But will fix it later today.

kennethjiang, 2024-01-11:
@Guilouz Working on this now. but I realized that `[include printer_params.cfg]` is in `printer.cfg` only when a user has installed using the helper script.

Since the macro file is intended for all moonraker-obico users, including non-K1 users - any idea about a more general approach that works for most or all klipper scenarios??

kennethjiang, 2024-01-11:
Will the `[printer]` line always be there? if so, I can add in front of that line.

kennethjiang, 2024-01-11:
I don't like this approach. It looks like `[printer]` usually appears in the middle of the cfg file and hence it's ugly to insert there.

I think a better approach is to insert it before the first `<---------------------- SAVE_CONFIG ---------------------->` line. And if one doesn't exists, insert it at the bottom of the file.

I implemented it here: https://github.com/TheSpaghettiDetective/moonraker-obico/commit/1cab320022f45773395b8a653eb6ffe8d4a21b0f

Not the prettiest code I have written. Feedback/comment welcome! @Guilouz

Guilouz, 2024-01-11:
@kennethjiang [include printer_params.cfg] is a config file included in K1 config.

Maybe you can add a condition like this to determine if it's K1 OS or not, the K1 and K1 Max run an OS called Buildroot. We detect that by looking at the os-release file.:
```
if grep -Fqs "ID=buildroot" /etc/os-release
```

kennethjiang, 2024-01-11:
> 
> ```
> if grep -Fqs "ID=buildroot" /etc/os-release
> ```
I'm a little bit afraid that Creality will change this in the future... 

I guess there is no perfectly clean solution here. I'll just go with https://github.com/TheSpaghettiDetective/moonraker-obico/commit/1cab320022f45773395b8a653eb6ffe8d4a21b0f unless you have any objections @Guilouz . Will release it with the next update.

Again I appreciate your help! @Guilouz

Guilouz, 2024-01-11:
@kennethjiang ok no problem the main thing being that it is no longer at the start of the file.

I take care of removing these includes via my uninstallation script.

kennethjiang, 2024-01-12:
> @kennethjiang ok no problem the main thing being that it is no longer at the start of the file.
> 
> I take care of removing these includes via my uninstallation script.

Sounds good. Thanks again!

Guilouz, 2024-01-12:
Fixed with latest script update.

closed 2024-01-12 by Guilouz