Guilouz/Creality-Helper-Script-Wiki #264: [BUG] Mobileraker plugin not showing installed in app with newest firmware

riktanius, 2024-01-10:
### What happened?

Using Mobileraker, after running the script it still doesn't show in the app that the companion is working. Was working before, now not getting notifications. 

### What did you expect to happen instead?

I installed the companion using the script, opened the app, checked the status and shows not installed:

[image]

### How to reproduce this issue?

Do a full reset of the printer, upgrade the firmware to 1.3.3.3 and then use the script to install the mobileraker companion. Then open the app and check app settings

### Additional information:

Tried reinstalling and it didn't help.

Guilouz, 2024-01-10:
Can't reproduce issue. Mobileraker notifications works fine with latest firmware 1.3.3.3.

Factory reset must be done after update not before.

[image]

[image]

riktanius, 2024-01-10:
I thought the factory reset would take me back to the original Firmware. I will do the dance again and check back in.

closed 2024-01-11 by Guilouz