Guilouz/Creality-Helper-Script-Wiki #274: [BUG] Empty folder local using touch screen. Even stock files are missing after upgrade to V1.3.3.3

joergensen70, 2024-01-13:
### What happened?

Using the touch screen used to show the uploaded model files and the stock files like the benchy. 
Also downloading from orca won't bring the models into the local folder.
History is working fine and after print the models show up here.

### What did you expect to happen instead?

Would expect the stock model like the benchy on the local folder and also uploaded files

### How to reproduce this issue?

Install V1.3.3.3 and look into the local folder using the touch

### Additional information:

Updated from V1.3.2.31 to V1.3.3.3. In V1.3.2.31 that was working as expected

Guilouz, 2024-01-13:
I don’t understand what you mean

joergensen70, 2024-01-13:
[image]
I was talking about this screen. It was empty for me after the update. I tried a model from this screen but no model on the display. Local and USB was empty, history ok.
Now I try to take a picture and everything is ok again. Not sure what happened but the problem seems to be gone. Maybe you can just close the issue

Again, absolute awesome work you did. There are so nice features. The installation was absolutely great and I love the input shaper included graphs and the screw tuning.
Thank you so much for all your efforts

Guilouz, 2024-01-13:
don't forget that version 1.3.3.3 is a pre-release and some bugs may appear.

closed 2024-01-13 by Guilouz

joergensen70, 2024-01-13:
Yes, but I thought it might be helpful for you if you get aware of issues. So far I didn't have any other problems, other than the cam settings are not working. But I believe this is a hardware dependency