Guilouz/Creality-Helper-Script-Wiki #275: Probleme mit Timelapse [BUG] <title>

Schulli73, 2024-01-13:
### What happened?

Ich habe folgenden Fehler mit Timelapse: Ich habe Timelapse nach Anweisung vollständig installiert. 
Am PC kann ich die Webcam ganz normal sehen, aber an meinem iPad Pro und IPhone sehe ich nur ein weißes Bild. 
Unter den Einstellungen bei Timelapse kann ich keine Kamera auswählen, um Timelapse vollständig einzurichten. 

Ich habe die neueste Scriptversion 4.2 und die Firmwareversion 1.3.3.3  

### What did you expect to happen instead?

.

### How to reproduce this issue?

.

Guilouz, 2024-01-13:
In English please.

Schulli73, 2024-01-13:
> Bitte auf Englisch.

I have the following error with Timelapse: I have fully installed Timelapse as instructed.
I can see the webcam normally on my PC, but on my iPad Pro and iPhone I only see a white image.
Under the Timelapse settings, I cannot select a camera to fully set up Timelapse.

I have the latest script version 4.2 and firmware version 1.3.3.3

Guilouz, 2024-01-14:
Timelapse is a component to create timelapse of 3D prints.

Your issue have nothing to see with Timelapse but from what I understand it is the display of your webcam which works on PC but not elsewhere, right?

Schulli73, 2024-01-14:
hallo

As you can see in the first picture here, I see the image from the webcam quite normally on the PC.
[image]

As you can see in image 2, I cannot select a webcam under the settings on Timelapse to configure it correctly.
[image]

Picture 3 is a screenshot from my iPhone, where I only see a white image when I go to the webcam.
[image]

I hope the pictures help to understand the problem

Guilouz, 2024-01-14:
So nothing to see with Timelapse you have not image on your phone. It’s not related to wiki issue.
It’s normal tou can’t select camera on maisail it use camera by default.

Schulli73, 2024-01-14:
But if I add a webcam to the settings myself, I can also see it on my iPhone
[image]

Schulli73, 2024-01-14:
> Mit Timelapse gibt es also nichts zu sehen, was Sie nicht auf Ihrem Handy haben. Es hat nichts mit dem Wiki-Problem zu tun. Es ist normal, dass Sie die Kamera auf dem Maisail nicht auswählen können, es verwendet standardmäßig die Kamera.

 OK, thank you for the information

Guilouz, 2024-01-14:
Timelapse does not add a camera, it uses the camera already configured.

Schulli73, 2024-01-14:
> Timelapse fügt keine Kamera hinzu, sondern verwendet die bereits konfigurierte Kamera.

I tried it once

Guilouz, 2024-01-17:
Is it ok ?

Schulli73, 2024-01-17:
Hello Guilouz

Yes, the Webcam is working now, great work

Schulli73, 2024-01-17:
Hello Guilouz

Yes, the webcam is working now, great work

Schulli73, 2024-01-17:
hello

unfortunately not,
only when I create a web myself

Guilouz, 2024-01-17:
As I already said Timelapse have nothing to see with webcam display. 

Just try to remove webcam config and re-add webcam.

Schulli73, 2024-01-18:
Hello, I reset the printer again today and redid everything again. I can see the webcam on the original interface but still not on mainsail

[image]
[image]
[image]
[image]

cinealfa, 2024-01-20:
I am having the same issue, If I add the camera in moonraker I get a white screen. If change the port from 808 0 to 4409 I get the same white screen.Iff I comment the lines out in moo raker I can set up the camera in the settings but I cannot select it in the timelapse settings. Are you saying that it is not needed because timelapse will default to the only camera available?

avxmw, 2024-01-22:
I had a similar problem not having a camera view in Orca Slicer, solution/workaround for that was setting `service: mjpegstreamer-adaptive` - maybe that also works here.

Biospargel, 2024-01-23:
I also can't add the camera to the time lapses in the time lapse option, it is grayed out. otherwise it works in moonraker
[image]

Tombraider2006, 2024-01-23:
I think I know why all these users are having this problem. here is her solution:

1. go to the helper script and remove  the camera settings item
2. then go to moonraker.cfg and delete the [webcam camera] section. reboot, 
3. add the camera in the settings and then in the timelapse settings. everything will work. 

Now the question is:  why is this so? i don't know but it work.

Guilouz, 2024-01-23:
Strange issue. I use camera in moonraker.conf without issue.

closed 2024-01-23 by Guilouz

hjkeller, 2024-01-24:
I am having a similar issue. at first reboot and before I start a print, the camera works fine. Eventually during a print the camera will stop streaming. Using mainsail or mobileraker companion. Running 1.3.3.5 firmware, freshly reset and installed.

jeracom, 2024-02-07:
> I am having a similar issue. at first reboot and before I start a print, the camera works fine. Eventually during a print the camera will stop streaming. Using mainsail or mobileraker companion. Running 1.3.3.5 firmware, freshly reset and installed.

I have the same problem!