Guilouz/Creality-Helper-Script-Wiki #279: [BUG] <Timelapse does not create videos. missing ffmpeg>

GGianpaolo, 2024-01-14:
### What happened?

Good evening, yesterday I successfully unlocked my K1 with the help of these instructions Klipper/Fluidd.

Unfortunately, the creation of timelapse videos does not work. The camera works, images are also created but nothing happens.

I have set in OrcaSlicer that it should always render at the end of the print (and take one picture per layer change, according to instructions from Vez3D Youtube). I have also tried the macros, I can see in the console that the command has been sent, but unfortunately a video is not created.

If I go to Timelapse, I can still view the many images in the preview and I can also download the images as a zip file. However, when I click on Render, nothing happens.

Now I looked at the moonraker log and saw the following message:

---------------------moonraker.log snipping
2024-01-14 18:03:05,723 [timelapse.py:render()] - timelapse: /opt/bin/ffmpeg not found, please install ffmpeg
2024-01-14 18:03:05,723 [timelapse.py:render()] - /opt/bin/ffmpeg not found, please install ffmpeg
2024-01-14 18:05:25,370 [timelapse.py:render()] - timelapse: /opt/bin/ffmpeg not found, please install ffmpeg
2024-01-14 18:05:25,371 [timelapse.py:render()] - /opt/bin/ffmpeg not found, please install ffmpeg
2024-01-14 18:07:22,534 [timelapse.py:saveFramesZip()] - saved frames: timelapse_temperature_tower_PLA_57m15s.gcode_14-01-2024_18h06_frames.zip
---------------------moonraker.log 

However, I do not want to exclude the possibility of errors on my part.

### What did you expect to happen instead?

That i get a video after my print, just this cool function

### How to reproduce this issue?

So it doesn't work for me and is therefore reproducible, so I can provide logs, settings and other information from my system if required.

### Additional information:

Here is some information about my config's:

Printer.cfg
[include Helper-Script/timelapse.cfg]

moonraker.conf
[timelapse]
output_path: /usr/data/printer_data/timelapse/
frame_path: /usr/data/printer_data/frames/
ffmpeg_binary_path: /opt/bin/ffmpeg
snapshoturl: http://localhost:8080/?action=snapshot

/Helper-Script/timelaps.cfg
[gcode_macro TIMELAPSE_RENDER]
description: Render Timelapse video and wait for the result
variable_render: False
variable_run_identifier: 0
gcode:
  {action_respond_info("Timelapse: Rendering started")}
  {action_call_remote_method("timelapse_render", byrendermacro="True")}
  SET_GCODE_VARIABLE MACRO=TIMELAPSE_RENDER VARIABLE=render VALUE=True
  {printer.configfile.settings['gcode_macro pause'].rename_existing} ; execute the klipper PAUSE command
  UPDATE_DELAYED_GCODE ID=_WAIT_TIMELAPSE_RENDER DURATION=0.5

[image]
[image]
[image]
[image]

Guilouz, 2024-01-14:
Have you removed and reinstalled timelapse? It’s install updated ffmpeg

GGianpaolo, 2024-01-14:
Yes I tried that yesterday, when I do that it tells me that timelapse is successfully uninstalled, but then I can't install it again because I get the message that timelapse is already installed, even though it's not. 

So because of this I reinstalled everything yesterday because I was hoping that everything would work correctly afterwards.

if it can't be avoided then of course I'll do it again... 

is it possible to install ffmpeg without having to uninstall timelapse?

Guilouz, 2024-01-14:
Yes you can use opkg upgrade ffmeg in ssh

mattskent, 2024-01-14:
> when I do that it tells me that timelapse is successfully uninstalled, but then I can't install it again because I get the message that timelapse is already installed, even though it's not.

I had the same issue and was able to resolve by manually uncommenting the timelapse lines in moonraker.conf file and restarting.

Guilouz, 2024-01-14:
Nothing have changes in moonraker installation except now it's install updated ffmpeg with entware.

You must remove Timelapse and reinstall it and all is working fine, this have been tested.

GGianpaolo, 2024-01-14:
So, i have uninstalled timelapse with the helper, when i try to install it now, it tells me again that timelapse is already installed.

in the moonraker.conf timelapse is commented out. i was hoping that this might be the reason as mattskent says. unfortunately i can't get timelapse installed anymore.

Any idea how I can fix this now without resetting my system and reinstalling from scratch?

[ssh_log.txt](https://github.com/Guilouz/Creality-K1-and-K1-Max/files/13933103/ssh_log.txt)

mattskent, 2024-01-14:
My log looks exactly the same as above, only v1025 instead of 1023. Removing timelapse and reinstalling it with the script doesn't seem to be an option. Not sure how it got in this state.

@GGianpaolo if the timelapse lines are commented out, you'll want to UNcomment them and that should do it (at least that's what worked for me)

GGianpaolo, 2024-01-14:
ah ok, i will test ist now... thx

Another picture from update checks... seems like the uninstallation did not go correctly, since the rest is still there or am I misunderstanding...

[image]

GGianpaolo, 2024-01-14:
@mattskent 
i activated the timelapse lines in moonraker.conf, i.e. removed #. then i tried to install, but i still get the same message that timelapse is already installed.

unfortunately your workaround doesn't seem to work for me.

Guilouz, 2024-01-14:
My bad, i have missed uninstall component.

In SSH enter this commands:

rm -f /usr/data/moonraker/moonraker/moonraker/components/timelapse.py
rm -f /usr/data/moonraker/moonraker/moonraker/components/timelapse.pyc

Update script to latest version and reinstall it.

GGianpaolo, 2024-01-14:
@Guilouz 

all good, thank you very much for your efforts! 

one more question, how can I update the script? 

should I enter this command: 
`cd && rm -f installer.sh && wget --no-check-certificate https://raw.githubusercontent.com/Guilouz/Creality-K1-and-K1-Max/main/Scripts/installer.sh`

thanks for all

GGianpaolo, 2024-01-14:
I have successfully updated the script to 4.2.1

I am very grateful to you and also to mattskent for the support!

I was able to install timelapse again, will test it and write you a result here.

GGianpaolo, 2024-01-14:
Timelapse now also works.

Thanks again!

Guilouz, 2024-01-15:
Yes actually display on web interface diesn't work because I use mjpeg encoding for better and faster video encoding.
h264 works but encoding is too long.

closed 2024-01-15 by Guilouz

Guilouz, 2024-01-15:
Fixed. Please update latest script version, remove and reinstall Timelapse.

PospiLEE, 2024-01-15:
Hi. I have the same problem. I was remove timelaps moonraker. Deleted all configuration and once again install the timelaps moonraker. The all is correctli installed and now I have a ffmpeg. Problem is with setup on the Settings-Timelaps the Camera "select which camera will be used" is everytime disabled and I have no chance open and select the Camera. On the Settings-Webcams I have configured the cameras, and on the moonraker.cfg to.
[image]

GGianpaolo, 2024-01-15:
@PospiLEE 

have you also updated the installation helper to the latest version?

so everything works for me now.

i would recommend looking at the log of moonraker, maybe you can see there why you have this problem.

PospiLEE, 2024-01-16:
The last helper is with number 4.2.1. On the log file I looked but I do not see anything wrong, but I looken on this file a long time, then maybye I anythyng do not see.  PLS look maybye you see anything what I something  overlooking.

[image]
[image]
 
[moonraker.log](https://github.com/Guilouz/Creality-K1-and-K1-Max/files/13946458/moonraker.log)

GGianpaolo, 2024-01-16:
@PospiLEE 

Just for your information, I'm not a pro in the whole 3D world myself, I've only recently started using it and have only had my K1 for a few days. 

In your log there is a reference to a faulty repo of the Moonraker component... 

I would recommend you uninstall moonraker-timelapse again, restart the system, then reinstall moonraker-timelapse
[image]

PospiLEE, 2024-01-16:
Ok, then I was doing, but not helped me. I stay on the same problem :-(.   On the log file you found a "untracked source file" do you know where can i verify the route? When I looked on the way, then I think that I have too many folders to watch timelaps.py. But I do not know when is this way correct.

[image]

GGianpaolo, 2024-01-17:
@PospiLEE 

Please give me your email address or your Discord name and I will contact you via that. This error here has ended successfully so far. This way we don't spam this thread.

PospiLEE, 2024-01-17:
Thx so much 3dprint@abc-it.cz. 

> @PospiLEE
> 
> Please give me your email address or your Discord name and I will contact you via that. This error here has ended successfully so far. This way we don't spam this thread.