Guilouz/Creality-Helper-Script-Wiki #284: [BUG] Section 'gcode_shell_command resonance_graph' is not a valid config section

Borega, 2024-01-15:
### What happened?

After install of the improved shapers calibrations I now get the following error:
Section 'gcode_shell_command resonance_graph' is not a valid config section
Klippy.log shows the following:
[ERROR] 2024-01-15 20:04:37,956 [root] [klippy:_connect:201] {"code":"key341", "msg":"Section 'gcode_shell_command resonance_graph' is not a valid config section", "values":["gcode_shell_command resonance_graph"]}
[ERROR] 2024-01-15 20:04:37,959 [root] [klippy:_connect:231] Config error
Traceback (most recent call last):
  File "/usr/share/klipper/klippy/klippy.py", line 193, in _connect
    self._read_config()
  File "/usr/share/klipper/klippy/klippy.py", line 163, in _read_config
    pconfig.check_unused_options(config)
  File "/usr/share/klipper/klippy/configfile.py", line 304, in check_unused_options
    raise error("""{"code":"key341", "msg":"Section '%s' is not a valid config section", "values":["%s"]}""" % (section, section))
configparser.Error: {"code":"key341", "msg":"Section 'gcode_shell_command resonance_graph' is not a valid config section", "values":["gcode_shell_command resonance_graph"]}

Unit is a K1 with 1.3.2.20 firmware.
Removing the improved shapers calibrations removes the error. 

### What did you expect to happen instead?

Normal boot behaviour as with all the other installs via the script

### How to reproduce this issue?

install the improved shapers calibrations

center-777, 2024-01-15:
I have the same problem.
When reinstalling "Klipper Gcode Shell Command", I get the following error

[image]

From "System" menu:
Disk Usage: 1.2G / 6.5G (5.0G available)

Borega, 2024-01-15:
commenting out the section in shapers-configs.cfg results in the error moving the belts_graph (as I  had expected). Seems to be a reference problem or maybe a problem of missing py packages?

Borega, 2024-01-15:
The issue was resolved by uninstalling Klipper Gcode Shell Command that I had installed from an old version and installing it anew. Then install input shaper fix and now it works

closed 2024-01-15 by Borega