Guilouz/Creality-Helper-Script-Wiki #288: [BUG] INSTALLER.SH fails to run or update

Ltek, 2024-01-17:
### What happened?

```
root@K1Max-53C2 /root [#] cd && sh ./installer.sh
./installer.sh: line 2007: syntax error: unexpected end of file (expecting "done")
```
So I deleted INSTALLER.SH and tried to reinstall. It will not run (no idea how/where i'm out of space - I just ran prints fine 2 days straight. Power off/on didnt help..

```
Connecting to raw.githubusercontent.com (185.199.109.133:443)
saving to 'installer.sh'
installer.sh          16% |******                               | 36864  0:00:05 ETA
wget: write error: No space left on device
wget: write error: Connection reset by peer
```
I cleared the swap file and checked memory...

```
root@K1Max-53C2 /root [#] free -m
              total        used        free      shared  buff/cache   available
Mem:            197         114           3           3          79          78
Swap:             0           0           0
```
then tried again...

```
root@K1Max-53C2 /root [#] cd && sh ./installer.sh
./installer.sh: line 2007: syntax error: unexpected end of file (expecting "done")
root@K1Max-53C2 /root [#] cd && wget --no-check-certificate https://raw.githubusercon
tent.com/Guilouz/Creality-K1-and-K1-Max/main/Scripts/installer.sh
Connecting to raw.githubusercontent.com (185.199.108.133:443)
wget: can't open 'installer.sh': File exists
```

### What did you expect to happen instead?

be able to run or update

I think this bug is the the Installer script... I'm not out of space :: since I found this bug I've ran 10+ prints and I just 5 mins ago updated Moonraker via Fluidd.

### How to reproduce this issue?

no idea but at least one other has reported they have the same problem

https://github.com/Guilouz/Creality-K1-and-K1-Max/discussions/278

Ltek, 2024-01-17:
update... looks like .cache was filling up the space INSTALLER uses. No idea why it was full but it stopped Installer.sh from running.

Guilouz, 2024-01-17:
That's mean your root folder is full.

Just use this command: rm -rf /root/.cache

closed 2024-01-17 by Guilouz