Guilouz/Creality-Helper-Script-Wiki #289: [BUG] Klipper Adaptive Meshing & Purging mechanical crash

TheTesla, 2024-01-17:
### What happened?

If a 360 mm long and 60 mm wide object is fitted 45° diagonally between opposite corners of the print bed, crossing its center on the Creality k1 max, it purges a drop instead of a line in the center of the bed. While printing the first layer, the print head crashes into this drop.

### What did you expect to happen instead?

creating a 45° purge line near the object

### How to reproduce this issue?

print an object with 360 mm by 60 mm footprint, 45° diagonally between two opposite corners

### Additional information:

[image]
[image]

Guilouz, 2024-01-17:
Not possible to do that.
When purge line can’t be done in front of model a purge blob is done on the center and must be removed before start print.

closed 2024-01-17 by Guilouz

itay-grudev, 2024-01-19:
Can it be done in a corner or on the back of the build plate?
Is that configurable?
Is the setting in another project?