Guilouz/Creality-Helper-Script-Wiki #294: Cant restore Creality Web Interface

duttyrock, 2024-01-19:
### What happened?

Hello,

First of all, I would like to thank you for your great work.

I wanted to upgrade to FW 1.3.3.3 today. Previously I removed GuppyScreen and wanted to restore the Creality Web Interface. According to the console, the web interface was successfully restored and the Creality Screen appeared, but I could not establish a browser connection to port 80. I installed GuppyScreen again and the message appeared:

Backing up original files...
mv: can't rename '/etc/init.d/S12boot_display': No such file or directory
cp: can't stat '/etc/init.d/S99start_app': No such file or directory

Then I noticed that after a restart, the Creality Screen no longer booted. I then reinstalled GuppyScreen and removed it again. Now I was back in the Creality Screen. Then I installed the update 1.3.3.3 via USB and again no boot from the Creality Screen. Fortunately, I was still able to establish the SSH connection and start your script. However, no current FW was specified under 7) System.

I then installed and removed GuppyScreen again and fortunately was back in the Creality Screen. I only had to confirm the FW update :D.

Now everything is back to the old state, with FW 1.3.3.3.

I have changed the hostname and the root password, otherwise I have done everything according to your instructions, so that is probably not the problem.

I noticed two other small points:

1. The system time or the time zone cannot be changed, so the log files are sometimes very difficult to evaluate. I have not found a quick way to do this under BusyBox...I miss timedatectl :)

2. The side fan does not regulate correctly in my System (Fluidd/Mainsail). 50% - 128/255 and the fan just hums. 80% - 204/255 and the fan hums and turns very slowly. However, when I print PLA with 80% side fan, the GuppyScreen shows 94%. 

Thanks again for your effort and have a nice weekend :)

### What did you expect to happen instead?

...

### How to reproduce this issue?

...

Guilouz, 2024-01-19:
Changing root password cause issue.
Restore factory settings.

duttyrock, 2024-01-19:
WTF :( Are you working on a solution?
Can I easily do a factory reset if I can't restore the Creality web interface?

Thank you.

Guilouz, 2024-01-19:
No you can’t. Restore settings from screen and do a factory reset

duttyrock, 2024-01-20:
Everything works fine.

Thank you.

closed 2024-01-20 by duttyrock