Guilouz/Creality-Helper-Script-Wiki #295: [BUG] Cannot install Octoeverywhere - No space left on device - K1 Max

TrainAss, 2024-01-19:
### What happened?

When attempting to install Octoeverywhere via the install script, after the install completes the pip stage and moves on to installing or updating python, message comes up "ERROR: Could not install packages due to an OSError: [Errno 28] No space left on device"

### What did you expect to happen instead?

OctoEverywhere to install

### How to reproduce this issue?

Running the installer script and installing everything but Fluidd, Entware, Improved Shapers, Mobileraker Companion, and Guppyscreen. When trying to install 'OctoEverywhere' after getting to the Python step, it fails stating no free space.

Guilouz, 2024-01-19:
That’s means you have no free space to install it.

try to reset cache from script or with this command: rm -rf /root/.cache

TrainAss, 2024-01-19:
Tried that a few times. No change. Still coming back saying there is no space left. It's only OctoEverywhere that does it though when installing the Python files which is the strange thing.

Guilouz, 2024-01-20:
Do factory reset and reinstall.

TrainAss, 2024-01-20:
This is a fresh install as of Tuesday. It's fine, I'll figure it out. Thanks though.

closed 2024-01-20 by TrainAss