Guilouz/Creality-Helper-Script-Wiki #300: [BUG]  python dependencies missing for input shaper

TheTesla, 2024-01-21:
### What happened?

Input shaper (with or without guppy screen) can't find numpy and fails.

### What did you expect to happen instead?

Input shaper should pass without errors.

### How to reproduce this issue?

Run input_shaper_calibration macro.

### Additional information:

numpy is missing in the `/usr/share/klippy-env`. Trying to install it ends in `[Errno 2] No such file or directory: '/opt/wuhui/code/creality/Pre-release/ingenic_linux_CR4CU220812S11/buildroot/buildroot/output/host/bin/mips-linux-gnu-gcc`

Guilouz, 2024-01-21:
I don’t know what you have changed but inout shaper is working.

restore to factory and reinstall

TheTesla, 2024-01-21:
How should I restore the factory settings?

Guilouz, 2024-01-21:
Read the wiki…

TheTesla, 2024-01-21:
OK, thanks. It works now, I restored to factory settings. Maybe I messed up the setup by installing byobu and tmux. I did everything in the byobu session...

closed 2024-01-22 by Guilouz