Guilouz/Creality-Helper-Script-Wiki #321: [BUG] Latest Installer Script not starting

Hammerfest, 2024-01-26:
### What happened?

Installer script is not running after updating the script when prompted for first time setup/install.

### What did you expect to happen instead?

Installer runs like it did on my old unit before return.

### How to reproduce this issue?

Following the guide here:
https://github.com/Guilouz/Creality-K1-and-K1-Max/wiki/Installation-Helper-Script

After I connected to SSH I ran the command to fetch/install the script to /root
I then ran the installer via the command line as I had done prior
The installer this time said there was an update, would I like to update, so I hit Y
It said successful/completed (ya I dont recall exact wording), press Enter to start installer script.
I hit enter

Nothing happens.

Hammerfest, 2024-01-26:
Despite it reporting the update was successful, the solution I found was to run the update command manually, then the script started when run.

SmooshWorkstations, 2024-01-26:
same issue here, waiting for a solid workaround.

SmooshWorkstations, 2024-01-26:
its working for me now, i reset my modem, reset my PC and M1 Max, ran the helper script in the following order

cd && rm -f installer.sh && wget --no-check-certificate https://raw.githubusercontent.com/Guilouz/Creality-K1-and-K1-Max/main/Scripts/installer.sh

enter, then

cd && sh ./installer.sh

now moonraker is downlaoding for me, (wasn't earlier today). I was also getting the prompt that there was an update, and if I wanted to DL it, i pressed yes, then it didnt run the updated script. I seem to not be getting that now.

Guilouz, 2024-01-26:
So it's ok ?

SmooshWorkstations, 2024-01-26:
It’s working now, nothing I did changed the fact the downloads were failing. I waited a day and it worked afterwards. Thanks for checking in. Kind Regards,

Hammerfest, 2024-01-26:
I did not have downloads failing, his issue is different from the one i experienced.
Its possible the "auto" update from the script failed, but still reported successful.

I dont have the time to factory reset my system to retry.

Like I said, the workaround I did was manually run the update then run the script when the script+auto update failed the first time.

SmooshWorkstations, 2024-01-27:
I was able to run the script,  got through about half the process then when I was trying to install fluid, it came up with a “download error” in the script. The latest script was working. Tried to install another tool, it said download failed.I tried all the usual troubleshooting like resetting firmware and re running the script. And nothing would install.I just waited until the next day, and it was fine. Problem isn’t there anymore.Kind Regards,

Hammerfest, 2024-01-27:
yes, your issue is different then the one I made this bug post for

Hammerfest, 2024-01-27:
As it turns out, I think I am just going to factory reset and try again.
KAMP is dumping its blob in the middle of my models despite label/exclude being selected in orca, camera controls return: 
$ CAM_AUTO_FOCUS
09:06:11 PM 
// Running Command {v4l2-ctl}...:
09:06:11 PM 
// unknown control 'focus_auto'
09:06:11 PM 
// Command {v4l2-ctl} finished

and so on for all camera commands, I did confirm it says installed in the installer, removed it and re-added it, still no joy

Hammerfest, 2024-01-27:
PLACEHOLDER (manged to get it to recurr, but didnt save the screenshot, FFS)

Guilouz, 2024-01-27:
You have something wrong with your permission i think. try to delete /tmp/curl file.

And I see script version is not displayed, that's mean you can reach url to check script.

Hammerfest, 2024-01-27:
> You have something wrong with your permission i think. try to delete /tmp/curl file.
> 
> And I see script version is not displayed, that's mean you can reach url to check script.

I think you may have missed the rest of what I said however, it doesn't ALWAYS do this, I ran it 10 times in a row and it happened only 3 times, SOMETHING is happening which causes it to choke and for some reason it skips to check for update.

Maybe a simple try again if permission denied situation is detected, like I said, I just ran the script again and it worked fine, so 
if encounter permission issue, quit and retry
something as simple as that (because thats all I did)

itay-grudev, 2024-01-28:
Just a note - after the issue with `/tmp/curl`, you will end up with an empty `installer.sh` file. You will need to re-download it.

Hammerfest, 2024-01-28:
> Just a note - after the issue with `/tmp/curl`, you will end up with an empty `installer.sh` file. You will need to re-download it.

that does explain why running it after did nothing and running the manual update fixed it

nirsinay, 2024-01-28:
check the date on the K1 using the date command. in my K1 it was set to march 1st 2020. after using date -s to update the k1 time the issue was resolved

Adur86, 2024-01-28:
Hello everybody. i am very new to all of this. Tried the installer script and have the same issues. I couldnt even install fluidd, so i took "creality-version" of it. i thought that running fluidd would help me with the installer-script. i can instasll entware, but installing kamp or gcode-shell-commands is not possible. Updating to the newer version of the script is also not possible. by typing "yes" the installer.sh is then empty (nothing in it!). i can get the script running by hitting "no" and enter. but then i cannot install the tools/updates/utilities. Something is wrong with the uptdater. Even updating manually does not help.. :( I would be very very happt to hear from you guys!! Thank you very much in advance
[image]

Guilouz, 2024-01-28:
Download latest version, I have added more checks for auto-update.

Hammerfest, 2024-01-28:
Ill try to test this out this evening if not it may not be until next weekend.
Thanks for all the time spent!

Adur86, 2024-01-28:
Hallo Guilouz,
i just tried but fail to open the Helper-Script (Screenshot below).
[image]

Adur86, 2024-01-29:
Just tried again. i was not able to run the update of the helper script via pressing y (after the installer.sh command). I then looked up the version of the helper script. It says 4.3.6 which is the latest version i guess???? I then tried installing mainsail, gcode shell commands and kamp. the download always failed due to "github-ssl-issues" (?). What can i do? Resetting the printer (via html-Script) and try again with the helper-script version 4.3.6??? 🙄 Thank you for your reply in advance!

Hammerfest, 2024-01-29:
> Just tried again. i was not able to run the update of the helper script via pressing y (after the installer.sh command). I then looked up the version of the helper script. It says 4.3.6 which is the latest version i guess???? I then tried installing mainsail, gcode shell commands and kamp. the download always failed due to "github-ssl-issues" (?). What can i do? Resetting the printer (via html-Script) and try again with the helper-script version 4.3.6??? 🙄 Thank you for your reply in advance!
> 
> Edit: It seems that the time and date is always set back and is not the actual time and date of today?! Could this info be important? Maybe somehow this causes the issues in downloading/installing (bug in firmware to prevent us all from installing fluidd/mainsail.. 😂 )

Your issue is different from the one outlined in this ticket, I recommend you make a new one.

This ticket issue is about getting permission denied on run yet asking to update anyway and it failing to update, yet saying its successful but leaving you with an empty installer script file.

Adur86, 2024-01-29:
> Your issue is different from the one outlined in this ticket, I recommend you make a new one.
>
> This ticket issue is about getting permission denied on run yet asking to update anyway and it failing [...]

No. You are Not right. I Had the same "Update -fail" AS you with the Installer being empty when updating. But with the Manual Update i found Out when your skipping to the "old-version" that current Version of 4.3.6 ist already installed. Please read my postings above. I think that the Skript or the installer.sh is "Brocken" in some ways. That is why i posted IT Here.

Hammerfest, 2024-01-29:
> No. You are Not right. I Had the same "Update -fail" AS you with the Installer being empty when updating. But with the Manual Update i found Out when your skipping to the "old-version" that current [...]

Yes, I read your message, this ticket is for the first part of the issue which is different from your issue, I never have had connection issues with github fetching files, this issue is specifically to deal with permission denied then fetching an empty file

You may have started with this issue, but your ending is with another issue

jrezin, 2024-01-29:
I've just factory reseted the my machine and reinstalled the script.
from memory (sorry no more details and, maybe, inaccuracies), what happened was :
On the first run, it "detected a new version" (even if it was just downloaded : v.4.3.6) and complained about no curl permission denied. [same message as @Hammerfest 's]
I answered `no` the the upgrade, and after the count down it started, no errors them.

DisorderlyDan, 2024-01-30:
With your update just now, looks like it broke something.

> root@K1-9F2B /root [#] cd && sh ./installer.sh
> 
>  A new script version is available!
> 
>  See changelog here: https://tinyurl.com/w7d9k5bt
> 
>  Do you want to update ? (y/n): y
> 
>  ✓ The script has been updated! Press Enter to start the new version!
> 
> 
> /root/installer.sh: line 970: syntax error: unexpected "else" (expecting ";;")
> root@K1-9F2B /root [#] cd && sh ./installer.sh
> ./installer.sh: line 970: syntax error: unexpected "else" (expecting ";;")
>

Guilouz, 2024-01-30:
It’s already fixed

Hammerfest, 2024-01-30:
> @Hammerfest Is it ok with latest update ?
I havent been able to get free time, i might not have free time until this weekend.

To prevent more issues being incorrectly stacked here you can go ahead and close it and I will simply reopen or comment on when I get a chance if it doesnt resolve.

Guilouz, 2024-01-30:
Ok. Normally it’s fixed.

closed 2024-01-30 by Guilouz

Adur86, 2024-01-31:
> Ok. Normally it’s fixed.

Yes it is! Thank you very much!!! I resetted to factory settings, then set the time and date and loaded the manual update for your script and it worked perfectly!!!!! 😊😊😊 you cannot imagine how frustrated i was about all this! Thank you!