Guilouz/Creality-Helper-Script-Wiki #331: [BUG] Camera Settings Control Broken

Hammerfest, 2024-01-27:
### What happened?

Issue can be seen here:
https://github.com/Guilouz/Creality-K1-and-K1-Max/discussions/200

Bottom line, camera controls are broken, some speculate maybe different hardware used by Creality, currently unproven.

03:58:00 AM $ CAM_AUTO_FOCUS
03:58:00 AM // Running Command {v4l2-ctl}...:
03:58:00 AM // unknown control 'focus_auto'
03:58:00 AM // Command {v4l2-ctl} finished
03:58:01 AM $ CAM_BACKLIGHT_COMPENSATION
03:58:01 AM // Running Command {v4l2-ctl}...:
03:58:01 AM // unknown control 'backlight_compensation'
03:58:01 AM // Command {v4l2-ctl} finished
03:58:02 AM $ CAM_BRIGHTNESS
03:58:02 AM // Running Command {v4l2-ctl}...:
03:58:02 AM // unknown control 'brightness'
03:58:02 AM // Command {v4l2-ctl} finished
03:58:03 AM $ CAM_CONTRAST
03:58:03 AM // Running Command {v4l2-ctl}...:
03:58:03 AM // unknown control 'contrast'
03:58:03 AM // Command {v4l2-ctl} finished
03:58:04 AM $ CAM_EXPOSURE_ABSOLUTE
03:58:04 AM // Running Command {v4l2-ctl}...:
03:58:04 AM // unknown control 'exposure_absolute'
03:58:04 AM // Command {v4l2-ctl} finished
03:58:05 AM $ CAM_EXPOSURE_AUTO
03:58:05 AM // Running Command {v4l2-ctl}...:
03:58:05 AM // unknown control 'exposure_auto'
03:58:05 AM // Command {v4l2-ctl} finished
03:58:07 AM $ CAM_EXPOSURE_AUTO_PRIORITY
03:58:07 AM // Running Command {v4l2-ctl}...:
03:58:07 AM // unknown control 'exposure_auto_priority'
03:58:07 AM // Command {v4l2-ctl} finished
03:58:08 AM $ CAM_FOCUS_ABSOLUTE
03:58:08 AM // Running Command {v4l2-ctl}...:
03:58:08 AM // unknown control 'focus_absolute'
03:58:08 AM // Command {v4l2-ctl} finished
03:58:09 AM $ CAM_GAIN
03:58:09 AM // Running Command {v4l2-ctl}...:
03:58:09 AM // unknown control 'gain'
03:58:09 AM // Command {v4l2-ctl} finished
03:58:10 AM $ CAM_GAMMA
03:58:10 AM // Running Command {v4l2-ctl}...:
03:58:10 AM // unknown control 'gamma'
03:58:11 AM // Command {v4l2-ctl} finished
03:58:12 AM $ CAM_HUE
03:58:12 AM // Running Command {v4l2-ctl}...:
03:58:12 AM // unknown control 'hue'
03:58:12 AM // Command {v4l2-ctl} finished
03:58:13 AM $ CAM_POWER_LINE_FREQUENCY
03:58:13 AM // Running Command {v4l2-ctl}...:
03:58:13 AM // unknown control 'power_line_frequency'
03:58:13 AM // Command {v4l2-ctl} finished
03:58:14 AM $ CAM_SATURATION
03:58:14 AM // Running Command {v4l2-ctl}...:
03:58:14 AM // unknown control 'saturation'
03:58:14 AM // Command {v4l2-ctl} finished
03:58:18 AM $ CAM_SETTINGS
03:58:18 AM // Running Command {v4l2-ctl}...:
03:58:18 AM // Command {v4l2-ctl} finished
03:58:20 AM $ CAM_SHARPNESS
03:58:20 AM // Running Command {v4l2-ctl}...:
03:58:20 AM // unknown control 'sharpness'
03:58:20 AM // Command {v4l2-ctl} finished
03:58:21 AM $ CAM_WHITE_BALANCE_TEMPERATURE
03:58:21 AM // Running Command {v4l2-ctl}...:
03:58:21 AM // unknown control 'white_balance_temperature'
03:58:21 AM // Command {v4l2-ctl} finished
03:58:22 AM $ CAM_WHITE_BALANCE_TEMPERATURE_AUTO
03:58:22 AM // Running Command {v4l2-ctl}...:
03:58:22 AM // unknown control 'white_balance_temperature_auto'
03:58:22 AM // Command {v4l2-ctl} finished

### What did you expect to happen instead?

Camera Macros Work

### How to reproduce this issue?

[image]
[image]

Guilouz, 2024-01-27:
What is the return of this command in SSH ?

v4l2-ctl -d /dev/video4 -l

JReacher2, 2024-01-27:
> What is the return of this command in SSH ?
> 
> v4l2-ctl -d /dev/video4 -l

i get no response from this command.

Guilouz, 2024-01-27:
What is the exactly return?

JReacher2, 2024-01-27:
> What is the exactly return?

[image]

Nastras, 2024-01-27:
I have put a second ai camera into operation on the K1 together with destinal. We had the same problem with a newer camera that did not accept video adjustment commands. It looks like creality has changed the hardware including the firmware.

The new ai camera is also no longer displayed as a Creality camera.

[image]

Hammerfest, 2024-01-27:
Thanks for confirming the hardware change at least.

From simply googling the new chip it should be more powerful at least spec wise, control wise is another story, someone with more know-how will have to probe that, but I am willing to help in anyway I can with testing.

Guilouz, 2024-01-27:
So I can't do anything about that unfortunately...

closed 2024-01-27 by Guilouz

Hammerfest, 2024-01-27:
> So I can't do anything about that unfortunately...

who made the original one? maybe we can go put a ticket over there

As for your script here, you could add a check and notify user it is incompatible

Guilouz, 2024-01-27:
Macro are mine but if new camera can't be controlled I can do nothing.
Checking already done in my latest script version just released.

Hammerfest, 2024-01-27:
> Macro are mine but if new camera can't be controlled I can do nothing. Checking already done in my latest script version just released.

Right so technically it can be resolved, someone just has to figure out the commands the new camera can be controlled with so it can be updated.

Guilouz, 2024-01-27:
the firmware of these cameras seems to prevent this

Hammerfest, 2024-01-27:
> the firmware of these cameras seems to prevent this

Yep, if there are any commands or troubleshooting that can be done to check what commands the firmware does accept please let me know.

In the meantime I have sent an email to cs@creality.com requesting a list of commands the camera will accept, OR a request to update the camera firmware to accept them again.

I highly recommend anyone with the new camera to do the same.