Guilouz/Creality-Helper-Script-Wiki #333: [BUG] <title>

Yukondokne, 2024-01-27:
### What happened?

When installing option 17 " 17) Install Moonraker Timelapse"
results:
 Type your choice and validate with Enter: 17

 ✗ Please install Entware first!

### What did you expect to happen instead?

Entware was installed. successfully

 ✓ Entware has been installed successfully!

still cannot install timelapse

### How to reproduce this issue?

just using the latest script as of 1-27-24

tested on both k1 and k1max same results

as296, 2024-01-27:
not install Moonraker Timelapse 
"Please install Entware first!"
https://github.com/Guilouz/Creality-K1-and-K1-Max/blob/296bf7d54b89d3ed84a5374db6102b917ed467a7/Scripts/installer.sh#L1734
elif [ ! -d "$entware_file" ]; then
replaced by
elif [ -d "$entware_file" ]; then
and work

bhast2, 2024-01-27:
> not install Moonraker Timelapse "Please install Entware first!"
> 
> https://github.com/Guilouz/Creality-K1-and-K1-Max/blob/296bf7d54b89d3ed84a5374db6102b917ed467a7/Scripts/installer.sh#L1734
> 
> 
> elif [ ! -d "$entware_file" ]; then
> replaced by
> elif [ -d "$entware_file" ]; then
> and work

This worked for me.

Guilouz, 2024-01-27:
My bad, I forgot to change, it checked for the presence of a folder instead of a file. this is fixed in the latest version

closed 2024-01-27 by Guilouz