Guilouz/Creality-Helper-Script-Wiki #336: [BUG] Folder permissions

giorgiomanieri, 2024-01-28:
### What happened?

KAMP folder is read-only and it's not possible to edit the cfg file

### What did you expect to happen instead?

Edit the kamp-setting file

### How to reproduce this issue?

Edit the cfg files inside Kamp folder

winbard, 2024-01-28:
HI,
you must use a terminal program for change.
MobaXterm e.x. like for the helper scipt
Than go out from the root.
And opben the folder usr/data/KAMP-for-K1-Series/Configuration
There is the file for change

giorgiomanieri, 2024-01-28:
> HI, you must use a terminal program for change. MobaXterm e.x. like for the helper scipt Than go out from the root. And open the folder usr/data/KAMP-for-K1-Series/Configuration There is the file for change

Sure, it's an option. But I preferred to replace the original folder with the same unrestricted one for faster Klipper editing. It would also be useful by default

Guilouz, 2024-01-28:
Tthis is normal, the KAMP files are sysmlinks from folder located in /usr/data to keep the files up to date when a kamp update is done. 
These files are not supposed to be modified except the configuration file. 
This is how the original version of KAMP works.
If you want to edit file you must copy it and include the new file in printer.cfg

closed 2024-01-28 by Guilouz