Guilouz/Creality-Helper-Script-Wiki #342: no access to screen menu and SSH

wa1vy, 2024-01-30:
### What happened?

Updated form rooted 1.3.2.31 to 1.3.3.5. Started the print, everything worked great. When the print ended, screen freezed at creality logo.

### What did you expect to happen instead?

 Fluidd is still working at 4408 port, can't access default creality web interface and can't access printer via SSH. So without both of them I can't reset or reflash the printer. Any ideas what to do?... Thank you in advance.

### How to reproduce this issue?

updating to 1.3.3.5

Guilouz, 2024-01-30:
Depending what you have installed. Have you perform a factory reset after update?

wa1vy, 2024-01-30:
I cannot reset, because the html for reset do not see ip. However, when I ping it in cmd, I get replies.
[image]

Guilouz, 2024-01-30:
> Is there a way for a force flash of the motherboard without accepting it on the screen etc?

Only with this process: https://github.com/CrealityOfficial/K1_Series_Annex/releases/tag/V1.0.0

What have you installed to have this ?

wa1vy, 2024-01-30:
https://drive.google.com/file/d/1T0TvtmtD22mEpzL5YJzbyJ7H-9QIoozB/view
that s the firmware I ve installed

Guilouz, 2024-01-30:
What else did you install?

tkunchick, 2024-02-01:
I have been having the same issue since updating to 1.3.3.5.

rebooting just has it display the creailty logo for a bit then goes blank.

One of the times I was sitting there when it happened.. it flashed something a few times that looked like it was saying that user had not accepted user agreement.. but it was flashing fast and hard to read before it went blank.

I think it has something to do with increasing the bed mesh count above 6x6, but not sure as I am still testing when I have time to play with it. I do know that 12x12 blanks the screen out after a few prints. 29x29 blanks the screen out really fast. Currently running an 11x8.

```
[bed_mesh]
speed: 150
mesh_min: 5,5
mesh_max: 295,295
probe_count: 11,8
algorithm: bicubic
bicubic_tension: 0.2
mesh_pps: 2, 2
fade_start: 2.0
fade_end: 20.0
move_check_distance: 5
split_delta_z: .025
```

might be some kind of memory issue? with the increased points? just guessing.. I have no real idea.  

But I reset everything Saturday, downgrade firmware, upgrade firmware, installed script, only installed from the script and it ran fine until I commented out all the `#tri_min` and `#tri_max`  and changed the` [prtouch_v2] pr_version: 1 `to increase probe count.

Myklt8, 2024-02-03:
I have the exact same issue, even after reverting back to version 2

Guilouz, 2024-02-03:
If you change probe_count in latest firmware, screen UI will not load. It's a limitation in prtouch_v2.

closed 2024-02-03 by Guilouz

tkunchick, 2024-02-03:
Is there somewhere i can read more about prtouch_v2? 

 My google skills seem lacking trying to find anything about it. 
 
Just from my experience, if you keep the count at 10 or leas the screen stays functioning.