Guilouz/Creality-Helper-Script-Wiki #344: Since last firmware update, impossible to start a shaper test, and impossible to get klipper connected on fluidd console.

philping, 2024-01-30:
### What happened?

Hi everyone ! On my K1 max (rooted), I decided to get the frech firmware update from Créality (1.3.3.5). 
Since this moment, impossible to start a new imput shaper test (when I press the "start test" icon,on the k1 max screen,  the sentense "test done" appears immédiately without any resonance test !
When I go on my computer, on  fluidd console, klipper is not connected and the message "Include file '/usr/data/printer_data/config/Helper-Script/camera-settings.cfg' does not exist".
I don't know what to do ....

### What did you expect to happen instead?

... a good klipper connection, and ability to get a proper resonnance test !

### How to reproduce this issue?

... how to not reproduce would be the best way ....

Guilouz, 2024-01-30:
Please post issue when it’s an real issue.

That’s mean you have not doing a factory reset and not reinstalled features after your update.

philping, 2024-01-31:
Sorry fot this ... I will delete this issue quickly !

closed 2024-01-31 by Guilouz